verilog-to-routing/vtr-verilog-to-routing #1410: Odin: wire is a net type and cannot be on left in always block

emacdo12, 2020-07-07:
#### Expected Behaviour

An error should occur when a net type is assigned a value in an always block. Only data types are allowed for that. ex reg

#### Current Behaviour

Odin does not error.

#### Possible Solution

#### Steps to Reproduce

```
module simple_op(a,b,c);

input a;
input b;

output wire c;

always @(*) begin
    c = a & b;
end

endmodule 
```

#### Context

Happened upon whilst creating keyword benchmarks

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

jeanlego, 2020-07-16:
@emacdo12 could you have a look into this? the error should be thrown during netlist creation. 
this would be a good place to start looking

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/ODIN_II/SRC/netlist_create_from_ast.cpp#L2969

sdamghan, 2021-02-02:
@harpreetbamrah would you please have a look?

sdamghan, 2022-05-03:
@christtc99 - Can you please have a look into this issue? We got the above PRs related to this issue, it is worth having a look at them to find out what is left to be done.

harpreetbamrah, 2022-05-03:
Thank you @sdamghan for pointing to this.
All the changes put up in the mentioned PR https://github.com/CAS-Atlantic/vtr-verilog-to-routing/pull/33 have been tested and work fine; it's just the expectation results that need to be regenerated, using may be `make regenerate_expectation` or `make test`, I'm not very sure about that.
Let me know if any further help required.

sdamghan, 2022-05-03:
Thanks, @harpreetbamrah - sounds like everything is already in place.
@christtc99 - then please update the branch of PR [CAS-Atlantic#33](https://github.com/CAS-Atlantic/vtr-verilog-to-routing/pull/33) with the current VTR master branch and regenerate the Odin-II expected results. To find out more about Odin-II regression tests and generating/regenerating expected results you would have a look at Odin-II documentation at [here](https://docs.verilogtorouting.org/en/latest/odin/dev_guide/regression_test/). The `make regenerate_expectation` command in this case would be good choice.