verilog-to-routing/vtr-verilog-to-routing #1430: Packer extensions

litghost, 2020-07-16:
#### Proposed Behaviour

In order to conform to some packing rules for the 7-series architecture, there are rules the pack needs to follow that I don't believe can easily be expressed.  I want to discuss them in turn, and determine good ways forward.

1. Mux priority

There are two cases in the 7-series CLB where the cluster router should prefer a particular mux path when possible over another.  In one case, the choice of mux path results in fewer connections to route by re-using an IPIN.  In the other case, the vendor tools simply requires one mux path to be prefered over the other, and we want VPR to follow it's rules to not create conflicts.

I propose creating a formal mux priority based on the order that inputs to a mux are defined.  This might regress to a depth first search, but given that cluster routing tends to be small, that might be acceptable?

2. Interior `blif_models`

So there are at least two cases where the 7-series CLB has "interior" blif_models.  What I mean by this is that the current leaf pb_type (say CARRY4) does have to modelled from a timing standpoint as a 4 tall carry chain.  However it can also be seen as containing 8 placeable primatives.  See below image:

[image]

I've highlighted the entire CARRY4, but inside you can see 4x MUXCY and 4x XORCY primitives that are also pack-able.

In theory you could split the CARRY4 in the 4x MUXCY and 4x XORCY primatives, however the timing paths are not cumulative.  So the CIN -> CO3 is not the delay you would get from stacking the MUXCY's.  So from a logical standpoint, the packer should have fine grained control over those MUXCY primatives, but the pb_type timing model precludes it.  Currently the 7-series VPR architecture only supports using the full CARRY4 primative, not the MUXCY / XORCY.

There is also the case of fused LUTs, where you have a LUT6_2, which is a 64-bit LUT, where there are two outputs, the full 64-bit LUT output (using A[5:0]) and the 32-bit LUT output (using A[4:0]).  The lower 32-bits of the LUT INIT controls the 32-bit output, and the full 64-bit LUT INIT controls the 64-bit output.  In case where the A[5] pin is tied high, this changes the LUT6_2 into two LUT5's (or smaller) as long as the inputs can be selected without conflict.  In the case where the A[5] pin is not tied high, if the LUT6 equation is the same as the LUT5 equation when A[5] is low, then they can be packed in the same primitive.

I will note that it is unclear how often it is needed to model the full LUT6_2.  Currently the 7-series VPR arch only supports either LUT6 or 2x LUT5.

#### Current Behaviour

#### Possible Solution

We are currently using 

#### Context

kmurray, 2020-07-18:
> I propose creating a formal mux priority based on the order that inputs to a mux are defined. This might regress to a depth first search, but given that cluster routing tends to be small, that might be acceptable?

Perhaps a more general way would be to add a cost (definable in the arch file) to each mux? That would guide the cluster router better guidance about what is prefered, and be a bit less mysterious than depending on declaration order in the arch file.

kmurray, 2020-07-18:
> I've highlighted the entire CARRY4, but inside you can see 4x MUXCY and 4x XORCY primitives that are also pack-able.

Perhaps I'm missing something, but wouldn't it be possible to model this as two different modes (one with the CARRY4, the other with the 4xMUXCY and 4xXORCY)? They could then have different timing models in that case.

> In theory you could split the CARRY4 in the 4x MUXCY and 4x XORCY primatives, however the timing paths are not cumulative. So the CIN -> CO3 is not the delay you would get from stacking the MUXCY's.

That seems rather unusual to me, if the underlying hardware is actually using the same MUXCY/XORCY primitives then things should add up (otherwise STA is broken as a general method...). One possibility is that the vendor simply made different timing modeling choices for these two scenarios. For instance produced a less pessimistic timing model for the carry4 (since you'd be averaging out some of the delay variations of the sub-blocks), but use a more pessimistic timing model for the MUXCY/XORCY.

litghost, 2020-07-20:
> if the underlying hardware is actually using the same MUXCY/XORCY primitives then things should add up (otherwise STA is broken as a general method...).

You are focusing the on logical expression of the hardware (4x MUXCY/XORCY), but not thinking about how the hardware designers could build a shorter logic path with a combined 4 tall carry chain.  Consider something like a https://en.wikipedia.org/wiki/Carry-lookahead_adder

The key idea here is that is possible to have hardware that must be model as a lumped whole for STA, but logically decomposes for clustering/placement purposes.

kmurray, 2020-07-21:
> You are focusing the on logical expression of the hardware (4x MUXCY/XORCY), but not thinking about how the hardware designers could build a shorter logic path with a combined 4 tall carry chain. Consider something like a https://en.wikipedia.org/wiki/Carry-lookahead_adder

I agree you could have a faster path if you had special hardware in parallel (and some configuration bit controlling a mux selecting between the fast carry and MUXCY/XOR paths etc.). But I thought you're concern was that the underlying hardware _was_ the same. My comments were that you could still end-up with different delays just depending on whether you modeled it as 4 independent MUXCY/XOR, or a single lumped CARRY4 (using the same hardware).

> The key idea here is that is possible to have hardware that must be model as a lumped whole for STA, but logically decomposes for clustering/placement purposes.

In the case where there is a configuration bit selecting between two underlying hardware paths, this sounds a lot like a mode.

From what you've described, does the vendor ever actually use a subset of the CARRY4 (e.g. 2 XOR, 2 MUX) but use the CARRY4 delay model? Or if they are using the sub-blocks do they just use a different delay model?

litghost, 2020-07-21:
> But I thought you're concern was that the underlying hardware _was_ the same.

The hardware _is_ the same.  However it can be understood and packed both as a 4 tall carry chain (CARRY4) and as 4x MUXCY / XORCY from a logical perspective.

> > The key idea here is that is possible to have hardware that must be model as a lumped whole for STA, but logically decomposes for clustering/placement purposes.
> 
> In the case where there is a configuration bit selecting between two underlying hardware paths, this sounds a lot like a mode.

There is not configuration bit, both the CARRY4 and 4x MUXCY / XORCY are the same hardware from a combinational logic perspective.  However if tried to decompose the delays from the CARRY4 into 4x MUXCY / XORCY, you would find it impossible, because the underlying hardware doesn't match the logic decomposition.