verilog-to-routing/vtr-verilog-to-routing #1434: Python linter in CI

jgoeders, 2020-07-16:
There are several Python scripts already part of the repo, and with #1429 beginning to convert the core scripts to Python, it would be good to run a python linter as part of CI. 

#### Proposed Behaviour
Run flake8 and/or pylint with CI checks.

#### Current Behaviour
No python linting in CI.

#### Possible Solution
Check output of flake8 and pylint with current checked-in python scripts to see where it stands currently.  May need to fix existing scripts to get the linter CI to pass.

jgoeders, 2020-07-19:
@mithro How did you envision us using the python linter?  I tried flake8 and pylint on some existing scripts and there are lots of issues that are detected, some more critical than others.  If we enforce a clean lint, it is going to take some work to clean up all of our scripts.

Are you using a python linter in other projects, and if so, how are they used?  

Some ideas:
- We could identify only certain things we want to look for (ie fail on uninitialized variable, but not on unused import)
- pylint provides a score out of 10 for each file.  We could enforce some minimum score.
- We could try and make all of the python files lint-free (would require larger effort)
- We could add an experimental travis CI check, and then over time have people clean up the python files (would be a good first commit task)

mithro, 2020-07-19:
@jgoeders -- I would probably start by enforcing new python files to be lint free and then slowly reduce the list of "grandfathered" existing Python scripts?

jgoeders, 2020-07-19:
OK, sounds good.  I'll start with a manual list files we are going to enforce, and we can slowly grow it until we have everything covered.

mithro, 2020-07-19:
@jgoeders I would suggest an **exclude** list (files we are not enforcing on) rather than an include list. That way any new files need to be compliant.

closed 2020-07-28 by jgoeders