verilog-to-routing/vtr-verilog-to-routing #1444: Extension of fc==0 logic

litghost, 2020-07-20:
#### Context

Currently I don't believe VPR can handle packing or placing nets that make use of both dedicated fabric ports (e.g. carry chains) and general routing fabric ports.

There are some behaviors in the packer that mean that it can make "impossible" choices.  The easiest example to visualize a carry chain that has a both a general routing fabric port and a dedicated path port.  The other common example that this arises in is IOB <-> IOI logic

#### Current Behaviour

Carry chains are implemented in narrow manor that doesn't generalize to non-fabric connections.

#### Proposed Behaviour

Generalize the packer and placer to support mixed dedicate and non-dedicated connections.

#### Possible Solution

I believe this issue can be addressed in a more general fashion if two problems can be addressed:

 1. The packer does not have awareness when routing to/from ports to restrict nets to ports that match their requirement
   - Example: If the packer routes a net to a fc==0 port, but the net is needs access to the general fabric, the router will fail
   - Example: If the packer routes a net to a fc!=0 port, but the net sources from a fc==0 port, the router will fail
   - Example: If the packer routes a net to a fc!=0 port, the carry chain will be implemented in general routing fabric resources, resulting in poor timing
 2. The placer doesn't know how to handle cases where a place macro net has more than 1 sinks https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/5c26d4d49809c10432a7e02d2d7b83f87c7db28f/vpr/src/place/place_macro.cpp#L135

I believe the solution to both 1 and 2 lies in splitting/filtering nets that connect to both fc==0 ports and fc!=0 ports.

The solution would require several changes.  

1. Determine which net sinks connect via fc==0 ports.  I believe this can be done during pre-packing, because carry chain patterns are built and annotated during pre-packing.
2. The packer would ensure that nets that have fc==0 and fc!=0 sinks are packed such that there are two OPINs for the net, one on a fc==0 and one on a fc!=0 port.  The packer can mark that the OPIN are equivalent for routing purposes via pb_type muxes (are a requirement).  This always the router to reach both fc==0 and fc!=0 sinks.  This can be done in a number of ways.
3. The placer would need to relax https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/5c26d4d49809c10432a7e02d2d7b83f87c7db28f/vpr/src/place/place_macro.cpp#L135 to verify that all fc==0 sinks have a satisfied macro

litghost, 2020-07-20:
@kmurray / @vaughnbetz Thoughts?

kmurray, 2020-07-21:
I recall some extended discussions with @mithro a while back on this topic. Its definitely an open issue.

In the long term, we hope to use the inter-block router to also route inside the clusters, which should avoid the awkwardness of two stage routing (which has issues like this one).

Your overall approach seems reasonable to me. As you noted we should be able to detect most of these cases pre-packing; although I expect there are some tricky corner cases.

One interesting thing to note is that currently the intra-cluster router assumes it can always loop-back (from an OPIN to an IPIN) by going outside the cluster to the inter-block routing (e.g. if it can't make an internal connection). You can see it being built here: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/lb_type_rr_graph.cpp#L237-L271. One potential idea is to modify the intra-cluster RR graph to encode the different possibilities (e.g. as different 'external' nodes which connect to different sets of pins). For instance, we shouldn't have an fc==0 pin connecting to the global routing 'node' (which may already be the case).

litghost, 2020-07-21:
> In the long term, we hope to use the inter-block router to also route inside the clusters, which should avoid the awkwardness of two stage routing (which has issues like this one).

This will not help because of output muxing congestion.  You would need to enable the router to pop blocks out of a cluster, because it will be possible to pack a cluster such that isn't routable.