verilog-to-routing/vtr-verilog-to-routing #1458: run_vtr_task script does not take the max_router_iteration setting in Titan task

tangxifan, 2020-07-31:
The `run_vtr_task.pl` script does not take the `max_router_iteration` setting in Titan task config file but give a random number.

#### Expected Behaviour

In the task config.txt file, user can specify the parameters passed to vpr run, e.g., 
`script_params=-starting_stage vpr --route_chan_width 300 --max_router_iterations 400 --router_lookahead map -timeout 172800`.
The run_vtr_task will pass the parameters when calling vpr. 
If we can the vpr.out log file, it should shows 
`/usr/bin/env time -v /research/ece/lnis/USERS/tang/research/EDA/vtr-verilog-to-routing/vpr/vpr stratixiv_arch.timing.xml stereo_vision_stratixiv_arch_timing --circuit_file stereo_vision_stratixiv_arch_timing.pre-vpr.blif --route_chan_width 300 --max_router_iterations 400 --router_lookahead map`

#### Current Behaviour

However, when running the titan benchmark, the vpr.out log file shows 
`/usr/bin/env time -v /research/ece/lnis/USERS/tang/research/EDA/vtr-verilog-to-routing/vpr/vpr stratixiv_arch.timing.xml stereo_vision_stratixiv_arch_timing --circuit_file stereo_vision_stratixiv_arch_timing.pre-vpr.blif --route_chan_width 300 --max_router_iterations 150 --router_lookahead map`

#### Possible Solution

Not sure if this problem occurs only to the titan benchmark configuration file.
Other configuration file may be impacted.

#### Steps to Reproduce

- Clone VTR repository `git clone https://github.com/verilog-to-routing/vtr-verilog-to-routing.git`
- Go to the repository and compile with `make -j`
- Fetch titan benchmark
- Copy the [titan_task_list.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5004577/titan_task_list.txt) to `vtr_flow/tasks/regression_tests/vtr_reg_weekly/`
- Run titan regression tests using `perl vtr_flow/scripts/run_vtr_task.pl -short_task_names -l ./vtr_flow/tasks/regression_tests/vtr_reg_weekly/titan_task_list.txt -j 23`

- Check the vpr.out file for any benchmark `vi vtr_flow/tasks/regression_tests/vtr_reg_weekly/vtr_reg_titan/latest/stratixiv_arch.timing.xml/sparcT1_core_stratixiv_arch_timing.blif/common/vpr.out`

1. 
2. 

#### Context

This is a functional problem of `run_vtr_task.pl`, which is detected by running the Titan benchmarks when applying RRGraph fix-up #1448 

#### Your Environment

* VTR revision used: latest master
* Operating System and version: CentOS 7.5
* Compiler version: g++-8.4

vaughnbetz, 2020-07-31:
Hi Jeff,

Xifan tells me this is holding up his work on re-tuning the Titan architecture; it's also going to hold up others in my group who are doing QoR tuning.  Xifan is running run_vtr_task.pl, not vtr_flow.py, so I'm not sure what's changed, but perhaps there is some link?

vaughnbetz, 2020-07-31:
@shadtorrie can you take a look?

shadtorrie, 2020-08-01:
This issue is caused by my changes with run_vtr_flow, I found what was causing it have have now fixed it. Should I make a new pull request to fix this issue?

vaughnbetz, 2020-08-01:
Yes, please do.  This would be good to get fixed right away.

shadtorrie, 2020-08-01:
@vaughnbetz I have submitted the pull request as seen above. @tangxifan could you review the fix as well?

tangxifan, 2020-08-01:
@shadtorrie I just tried your branch by running the titan benchmarks. The problem is fixed.

closed 2020-08-03 by tangxifan