verilog-to-routing/vtr-verilog-to-routing #1461: When building vtr , warning message: CMake Warning at libs/libeasygl/CMakeLists.txt:49 (message):   EasyGL: Failed to find required X11 library (on debian/ubuntu try 'sudo   apt-get install libx11-dev' to install)

anaarduengo, 2020-08-03:
#### Expected Behaviour

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

vaughnbetz, 2020-08-06:
Workaround: try the master branch; we've switched to ezgl (updated / more powerful rewrite of easygl layered over gtk).  Hopefully that builds.
Assigning to Mashad in case she has any ideas of why the required packages are not installing for 8.0.

shikc, 2022-08-21:
I have encountered the same problem and my solution is to change find_package(X11 COMPONENTS X11 Xft) to find_package(X11) in the file "vtr-verilog-to-routing-8.0.0/libs/libeasygl/CMakeLists.txt", line 38. Then, the VTR graphic can work properly.

closed 2025-05-25 by github-actions[bot]

vaughnbetz, 2025-05-25:
Resolved in 9.0 by switching to ezgl.