verilog-to-routing/vtr-verilog-to-routing #1470: Memory leaks in vtr_reg_strong graphics test

vaughnbetz, 2020-08-06:
When vpr is compiled with sanitizers (which check memory, much like valgrind), it has a few memory leaks when graphics are run.  This causes a regtest failure in vtr_reg_strong that should be fixed.

#### Expected Behaviour
vpr should not leak memory, or leaks that cannot be fixed (e.g. deep in a library that we can't fix) should be added to a supression file so that we don't flag them as errors.

#### Current Behaviour
See https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1459 for more details on how this was found and the full vpr.out.
The graphics_command regtest in vtr_reg_strong runs to completion, but has the following memory leaks:

=================================================================
==14302==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 768 byte(s) in 3 object(s) allocated from:
    #0 0x7fa2824ca602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x7fa27d1320b9  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x1d0b9)

Indirect leak of 160 byte(s) in 5 object(s) allocated from:
    #0 0x7fa2824ca79a in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x9879a)
    #1 0x7fa27d1327c8  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x1d7c8)

Indirect leak of 10 byte(s) in 2 object(s) allocated from:
    #0 0x7fa2824ca602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x7fa27ede4489 in __strdup (/lib/x86_64-linux-gnu/libc.so.6+0x8b489)

#### Possible Solution
The _strdup leaks are probably fixable.  The libfontconfig ones probably are not (this library is a known small leaker) but those errors should be added to the suppression file so they don't get flagged as failures in that case.

#### Steps to Reproduce
valgrind --leak-check=full
on this case may give a better trace of the routines called before the leaks.

The regtest and command line is (from #1459):

graphics_commands: k6_N10_mem32K_40nm.xml/stereovision3.v/common Error: Executable vpr failed

full command: /usr/bin/env time -v /tmpfs/src/github/vtr-verilog-to-routing/vpr/vpr k6_N10_mem32K_40nm.xml stereovision3 --circuit_file stereovision3.pre-vpr.blif --route_chan_width 100 --graphics_commands set_draw_block_outlines 0; set_draw_block_text 0; set_draw_block_internals 2; set_draw_net_max_fanout 128; save_graphics place.png; set_nets 1; save_graphics nets1.png; set_nets 2; save_graphics nets2.png; set_nets 0; set_cpd 1; save_graphics cpd1.png; --max_router_iterations 150

returncode : 23
log file : vpr.out
failed: Executable vpr failed (took 8.78 seconds)

#### Context
While the leaks are small, this is annoying since it causes CI failures, so we either need to fix it or turn off memory leak checking (which would be bad, as then leaks multiply!).

It is probably introduced in recent changes to graphics, so it may be in code you added Mahshad.

#### Your Environment
This was a recent VTR trunk build.

vaughnbetz, 2020-08-06:
This post might be helpful on sanitizers and finding the stack trace leading to a leaked memory allocation:
https://stackoverflow.com/questions/46575977/how-to-find-reason-of-memory-leak-with-leak-sanitizer

Seems if the libraries don't have debug symbols the sanitizers can't give a good stack trace; valgrind is worth trying in that case.
Also, Sameh knows how to update suppression files for sanitizers and valgrind.  As part of fixing this, it would be good if Sameh also added documentation on how to update these suppression files.

vaughnbetz, 2020-08-06:
@samehattia : Adding Sameh to comment on suppression files (and because he's tracked some libfontconfig leaks before).

vaughnbetz, 2020-08-08:
Sameh, I think you're the best person to track this one down and add it to a sanitizer file, so setting this to you.

samehattia, 2020-08-10:
Sounds good.

samehattia, 2020-08-10:
The sanitizer suppression file was added in 9c0353b (it's under vpr/lsan.supp), and it was included in the VPR run environment in 1c0818a. So as long as the executable is run through run_vtr_flow.pl, there shouldn't be any reported leaks related to libfontconfig.

I'll check if __strdup leaks are fixable or not.

samehattia, 2020-08-10:
I think the run_vtr_flow.py, which is currently used in running the regression tests instead of run_vtr_flow.pl, doesn't include the suppression files.

vaughnbetz, 2020-08-10:
@shadtorrie @jgoeders Shad, Jeff: can you take a look?  We should include the suppression files Sameh mentions with vtr_flow.py.

shadtorrie, 2020-08-13:
Regarding this issue, I cannot seem to reproduce it. I have added the suppression files and put in a pull request, I just am not  actually sure if it fixes the issue because I cannot reproduce the error.

closed 2020-08-14 by vaughnbetz