verilog-to-routing/vtr-verilog-to-routing #1471: Add doxygen link and doxygen info (comment style we want) to vtr developer guide

vaughnbetz, 2020-08-07:
vtr now builds doxygen documentation, but I don't know where the top level link is to get to it and I suspect others don't either.
We also don't tell developers that they should make new comments doxygen compatible.

#### Proposed Behaviour
1. doxygen should be built automatically or periodically and there should be a link to it from the VTR developer guide.  A good place would probably be right after Contrbution Guidelines in the Developer Guide:
https://docs.verilogtorouting.org/en/latest/dev/#

2. We should also have a section saying new comments should be formatted for Doxygen 
right after https://docs.verilogtorouting.org/en/latest/dev/developing/#code-formatting
 It should point out we're using Doxygen, give a few brief guidelines of what to do, and the link to the Doxygen website for more details.

#### Current Behaviour
We have doxygen, but most people won't know it or add to the doxygen commenting.

vaughnbetz, 2020-08-07:
@rw1nkler

rw1nkler, 2020-08-12:
> vtr now builds doxygen documentation, but I don't know where the top level link is to get to it and I suspect others don't either.
We also don't tell developers that they should make new comments doxygen compatible.
>
>Proposed Behaviour
>
> 1. doxygen should be built automatically or periodically and there should be a link to it from the VTR developer guide. A good place would probably be right after Contrbution Guidelines in the Developer Guide:
> https://docs.verilogtorouting.org/en/latest/dev/#

Doxygen is currently used for generating intermediate XML format, which is required by the Breathe plugin. Because of that, there is no "Doxygen HTML" output. The API documentation is built as a part of the VTR documentation in Sphinx, and is available here:
https://docs.verilogtorouting.org/en/latest/api/vpr/

The process of documenting the code is the following:

1. Document the code using Doxygen style comments
2. Create an RST file which uses Breathe directives to incorporate the code comments in the API documentation
3. Run Sphinx, which will convert the created RST file to HTML.

See: https://docs.verilogtorouting.org/en/latest/dev/c_api_doc/

> We should also have a section saying new comments should be formatted for Doxygen
right after https://docs.verilogtorouting.org/en/latest/dev/developing/#code-formatting
It should point out we're using Doxygen, give a few brief guidelines of what to do, and the link to the Doxygen website for more details.

I will prepare PR describing how the Doxygen comments should be applied to the code.

vaughnbetz, 2020-08-13:
Thanks; the documentation you pointed at looks good, and I think having guidelines on the comments we want in the future would be great.

I think it would be good to add a little more detail on how to document a new API.  Specifically, I think in this document https://docs.verilogtorouting.org/en/latest/dev/c_api_doc/
you should point the reader at an example path (e.g. https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/master/doc/src/api/vpr) and tell them how to document a new API / class.  I think this involves
(1) add a my_new_api.rst file
(2) add a new entry to index.rst

If you do this by showing an example, and link it to the 'how to write comments for doxygen' section (showing what comments are picked up by what doxygen command in the .rst file) it would make it easier for developers to see how to extend the documentation.

closed 2020-10-29 by vaughnbetz