verilog-to-routing/vtr-verilog-to-routing #1473: Travis CI intermittently failing to fetch llvm packages, causing fails

dpbaines, 2020-08-07:
#### Expected Behaviour

Travis CI normally tests pull requests to ensure they build correctly, and on many different compilers like GNU GCC and LLVM clang. 

#### Current Behaviour

On a number of PR's Travis is reporting failed jobs due to an inability to fetch LLVM libraries from apt-get. This doesn't happen always, and sometimes restarting the job fixes it, sometimes it causes another job to fail. Also it's not always the same libraries that are failing.

For example: https://travis-ci.com/github/verilog-to-routing/vtr-verilog-to-routing/jobs/369808911#L771 which is a part of my RCV branch merge PR. 

Also this https://travis-ci.com/github/verilog-to-routing/vtr-verilog-to-routing/jobs/369149961#L761

It is usually clang and clang format that fail. I've also encountered this failure: https://travis-ci.com/github/verilog-to-routing/vtr-verilog-to-routing/jobs/369808913
where apt-get is also unable to locate the clang package. 

This can be fixed by brute force by force restarting jobs.

#### Possible Solution

It could be an issue with the ubuntu repository mirrorlist, changing this to use a different mirror could present results. One temporary option is to remove the clang jobs for now. 

#### Steps to Reproduce

1. Create a pull request 
2. Observe CI test results

#### Context

It's effected two of my PR's so far, as well as another pending PR: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1466

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

vaughnbetz, 2020-08-08:
@litghost @HackerFoo any ideas?  This is annoying (slows down CI).  
We could try changing the mirror list as David suggests (best option), or could disable the llvm requiring CI (not too bad to turn off llvm compiler compatibility if it's causing these CI problems; turning off format checks would be a bigger deal).

HackerFoo, 2020-08-08:
It's possible to cache dependencies on Travis: https://docs.travis-ci.com/user/caching/

vaughnbetz, 2020-08-08:
That sounds like a good solution.

vaughnbetz, 2020-10-29:
This doesn't seem to be happening much anymore ... not sure if llvm.org got more reliable or we're caching.

PR #1608 (2021-01-14): ci: add GitHub Actions workflow 'Test', deprecate Travis CI
#### Description

This early work in progress for migrating from Travis CI to GitHub Actions. Not perfect yet, but it seems to work so far. I'm an absolute ignorant on the VTR codebase and on previous CI usage. I just dropped some hours here. Hence, any experienced user please review!

#### Related Issue

- Ref #1493
- Close #847
- Close #1473

#### Motivation and Context

/cc @mithro 

Discussion in verilator: https://github.com/verilator/verilator/issues/2658

#### How Has This Been Tested?

It's CI plumbing, so self-tested. See bel [... cut]
closed 2021-01-14 by litghost