verilog-to-routing/vtr-verilog-to-routing #1476: Sanitizer error in Threading Building Blocks

helen-23, 2020-08-08:
When running the titan_new DLA benchmark variants with the stratixiv_arch.timing.xml architecture under Titan flow, VPR succeeded, but the AddressSanitizer reported a few runtime errors in the TBB. There are also some memory leaks.

#### Expected Behaviour
Should not have any runtime errors with sanitizer build turned on.

#### Current Behaviour
The sanitizer currently reports the following 2 issues:
[image]
[image]

#### Possible Solution

- There are many discussions online about this issue. Most suggested that it is a defect in the TBB and that it could have been fixed by Intel in later versions, so updating the TBB in the future might work

- Perhaps turning threading off?

#### Steps to Reproduce

1. checkout my branch, **"vqm2bliff_one_lut_removal"**, which contains all changes required to run the DLA circuit
2. unzip the attached DLA circuit (DLA_BSC)
3. Run titan_flow.py with DLA_BSC and the SIV architecture capture. DO NOT run titan_flow.py with sanitizer build turned on, because there is currently integer overflow in the hash function due to multiply-add. Please do note to turn on options **--fit** and **--gen_post_fit_netlist**, because the DLA circuits need post-fit netlist for VPR. An example of the command looks like the following:

<titan_dir>/scripts/titan_flow.py \\
-q DLA_BSC/quartus2_proj/DLA.qpf \\
-a <vtr_root_dir>/vtr_flow/arch/stratixiv_arch.timing.xml \\
--fit \\
--gen_post_fit_netlist \\
--titan_dir <titan_dir> \\
--vqm2blif_dir <vtr_root_dir>/build/utils/vqm2blif \\
--quartus_dir /tools/intel/install/fpga/18.1/standard/quartus/bin \\
--vpr_dir <vtr_root_dir>/vpr

4. Now with sanitizer build turned on, run VPR with the generated post-fit BLIF and the provided vpr.sdc (DLA_BSC/run_flow/vpr.sdc). An example of the command looks like the following:

<vtr_root_dir>/vpr/vpr \\
<vtr_root_dir>/vtr_flow/arch/stratixiv_arch.timing.xml \\
DLA_stratixiv_post_fit.blif \\
--sdc_file DLA_BSC/run_flow/vpr.sdc \\
--route_chan_width 300 \\
--max_router_iterations 400 \\
--timing_analysis on \\
--timing_report_npaths 1000 \\

#### Context

#### Your Environment
* VTR revision used: 8.0
* Operating System and version: Linux Ubuntu 18.04.4 LTS (Bionic Beaver)
* Compiler version:

#### Attachments
[DLA_BSC (1).zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5044929/DLA_BSC.zip)

vaughnbetz, 2020-08-08:
Thanks Helen.  @kmurray, any ideas?  If this is a flaw in TBB, we could suppress the error.  Helen is running vpr without multi-threading on, so Tatum is presumably not talking to tbb much in this case either.

vaughnbetz, 2020-08-13:
This may be due to the lack of a sanitizer suppression file in CI (vtr_flow.py issue) which is tracked in another issue.

kmurray, 2020-09-03:
I've seen this before, it seems to be latent within TBB. It goes away if we build with out TBB, so I don't think its something we can fix. A suppression file is likely the best approach.

vaughnbetz, 2020-09-03:
Thanks Kevin.  It seems to have gone away once we fixed the suppression file. At least I haven't heard any more about it, which I'm interpreting as it is gone :).

closed 2020-10-29 by vaughnbetz