verilog-to-routing/vtr-verilog-to-routing #1505: Move travis build to xenial or newer as trusty has left supported status

mithro, 2020-08-26:
The travis build is currently using `dist: trusty`; see 
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/c5c1c0b59460e31f89e64d6da54b43c7b8b3cb41/.travis.yml#L4

The "End of Standard Support" for Ubuntu 14.04.6 LTS was April 2019 which has meant things like the llvm upstream packages have started to remove support. For example from https://apt.llvm.org/;
> Aug 20th 2019 - Ubuntu Trusty removed (EOL)

We should move to a new LTS version. The options are;
 - xenial - Ubuntu 16.04.7 LTS -- ends April 2021
 - bionic - Ubuntu 18.04.5 LTS -- ends April 2023
 - focal - Ubuntu 20.04.1 LTS -- ends April 2025

I recommend we move to focal as we won't be forced to update for 5 years :-)

mithro, 2020-08-26:
@vaughnbetz - Thoughts?

vaughnbetz, 2020-08-27:
Makes sense.  Is the reason the apt-get's have been problematic for llvm lately?
Not sure what needs to be done for this to happen, but sounds good to me.  From my side I'd appoint Mohamed as contact if we need to do things (once he's back from vacation). But I'm happy to let the experts in your group do this if at all possible!

PR #1511 (2020-08-28): travis: Upgrade OS to Bionic
This moves Travis to use `bionic` which is Ubuntu 18.04.5 LTS and has support until April 2023.

The "End of Standard Support" for Ubuntu 14.04.6 LTS was April 2019 which has meant things like the llvm upstream packages have started to remove support. For example from https://apt.llvm.org/;
> Aug 20th 2019 - Ubuntu Trusty removed (EOL)

Fixes #1505
closed 2020-08-28 by mithro