verilog-to-routing/vtr-verilog-to-routing #1513: Travis warning check failing due to gtk_widget_set_margin_start being deprecated

mithro, 2020-08-29:
With the update to Bionic, a new GTK version is installed and travis is now failing the "no warnings allowed" check as `gtk_widget_set_margin_start` has been deprecated.

See https://travis-ci.com/github/verilog-to-routing/vtr-verilog-to-routing/jobs/378950130
```
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/draw/draw_debug.cpp: In function ‘void draw_debug_window()’:
/home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/draw/draw_debug.cpp:47:9: error: ‘void gtk_widget_set_margin_left(GtkWidget*, gint)’ is deprecated: Use 'gtk_widget_set_margin_start' instead [-Werror=deprecated-declarations]
         gtk_widget_set_margin_left(mainGrid, 30);
         ^
In file included from /usr/include/gtk-3.0/gtk/gtkapplication.h:27:0,
                 from /usr/include/gtk-3.0/gtk/gtkwindow.h:33,
                 from /usr/include/gtk-3.0/gtk/gtkdialog.h:32,
                 from /usr/include/gtk-3.0/gtk/gtkaboutdialog.h:30,
                 from /usr/include/gtk-3.0/gtk/gtk.h:31,
                 from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/EXTERNAL/libezgl/include/ezgl/canvas.hpp:30,
                 from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/libs/EXTERNAL/libezgl/include/ezgl/application.hpp:22,
                 from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/draw/draw_debug.h:5,
                 from /home/travis/build/verilog-to-routing/vtr-verilog-to-routing/vpr/src/draw/draw_debug.cpp:1:
/usr/include/gtk-3.0/gtk/gtkwidget.h:1086:10: note: declared here
 void     gtk_widget_set_margin_left   (GtkWidget *widget,
          ^
```

mithro, 2020-08-29:
@MahshadJ  -- It looks like you wrote this code?

mithro, 2020-08-29:
https://developer.gnome.org/gtk3/stable/GtkWidget.html#gtk-widget-set-margin-left
> gtk_widget_set_margin_left has been deprecated since version 3.12 and should not be used in newly-written code.
> 
> Use gtk_widget_set_margin_start() instead.

MahshadJ, 2020-08-29:
I got those warnings when I ran my code locally as well, so changed all of them to the new name (gtk_widget_set_margin_start) but Travis CI would error and fail to build. This was around early July, has the GTK version been updated since?

PR #1514 (2020-08-29): vpr: Fixing the usage of deprecated GTK methods in drawing code.
Fixes #1513.

https://developer.gnome.org/gtk3/stable/GtkWidget.html#gtk-widget-set-margin-left
> gtk_widget_set_margin_left has been deprecated since version 3.12 and
> should not be used in newly-written code.
>
> Use gtk_widget_set_margin_start() instead.

https://developer.gnome.org/gtk3/stable/GtkWidget.html#gtk-widget-set-margin-right
> gtk_widget_set_margin_right has been deprecated since version 3.12 and
> should not be used in newly-written code.
>
> Use gtk_widget_set_margin_end() instead.
commit 304f548d (2020-08-29): vpr: Fixing the usage of deprecated GTK methods in drawing code.

Fixes #1513.

https://developer.gnome.org/gtk3/stable/GtkWidget.html#gtk-widget-set-margin-left
> gtk_widget_set_margin_left has been deprecated since version 3.12 and
> should not be used in newly-written code.
>
> Use gtk_widget_set_margin_start() instead.

https://developer.gnome.org/gtk3/stable/GtkWidget.html#gtk-widget-set-margin-right
> gtk_widget_set_margin_right has been deprecated since version 3.12 and
> should not be used in newly-written code.
>
> Use gtk_widget_set_margin_end() instead.
closed 2020-08-29 by mithro