verilog-to-routing/vtr-verilog-to-routing #1523: vtr_flow scripts search executable from system paths prior to ${VTR_ROOT} directory

tangxifan, 2020-09-03:
I am not sure if this should be considered a bug. Here is my context:
We installed a version of ABC in the system path, i.e., `/usr/local/bin/abc`, which is different than the version inside the vtr project.
When executing the vtr regression test script `./run_reg_test.pl vtr_reg_strong -show_failures -j20`, it reports errors like
```
Traceback (most recent call last):
  File "/research/ece/lnis/USERS/tang/github/vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_flow.py", line 647, in <module>
    retval = main()
  File "/research/ece/lnis/USERS/tang/github/vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_flow.py", line 376, in main
    return vtr_command_main(sys.argv[1:], prog=sys.argv[0])
  File "/research/ece/lnis/USERS/tang/github/vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_flow.py", line 435, in vtr_command_main
    check_place = args.check_place,
  File "/research/ece/lnis/USERS/tang/github/vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/flow.py", line 205, in run
    use_old_abc_script=use_old_abc_script,
  File "/research/ece/lnis/USERS/tang/github/vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/abc/abc.py", line 101, in run
    shutil.copyfile(str(abc_rc), str(temp_dir / "abc.rc"))
  File "/usr/lib64/python3.6/shutil.py", line 120, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/bin/abc.rc'
```
I checked the `abc.py` script, it seems that the vtr_flow scripts search executable from system paths prior to vtr_root directory.
```
    abc_exec = find_vtr_file("abc", is_executable=True) if abc_exec is None else abc_exec
    abc_rc = Path(abc_exec).parent / "abc.rc" if abc_rc is None else abc_rc
```

#### Expected Behaviour

The regression test script, as well as vtr_flow scripts, are supposed to run without these impacts from system paths.
The vtr_flow script should find executables first from the ${VTR_ROOT} directory rather than the system path.

#### Current Behaviour

The vtr_flow script spots an ABC executable from system path and most of the regression test failed in a local machine.

#### Possible Solution

Patch the vtr_flow script, maybe the `find_vtr_file()` function, to set priority in searching executables.

#### Steps to Reproduce

1.  Install an ABC to the system path, e.g., /usr/local/bin/
2.  Git clone the VTR repository
3.  Compile and run strong regression tests 
```
    ./run_reg_test.pl vtr_reg_strong -show_failures -j20
```

#### Context

#### Your Environment

* VTR revision used: 
Latest VTR master 
* Operating System and version:
CentOS 7
* Compiler version:
GCC-8.4

vaughnbetz, 2020-09-10:
Adding @sshadtorrie

closed 2020-09-11 by jgoeders

jgoeders, 2020-09-11:
Fixed with #1532