verilog-to-routing/vtr-verilog-to-routing #1527: Need better commenting of t_physical_tile_type and t_sub_tile

vaughnbetz, 2020-09-07:
Exactly how physical tiles are described, and what a sub_tile within each is, are core concepts.  Physical tiles were recently generalized to make it possible to place different types of logic blocks at one x,y location, and to allow multiple physical implementations of a logical block (e.g. CLB and MCLB).  However, the commenting of these types and how to use them isn't sufficient (or I'm not finding the big comment block). 

#### Proposed Behaviour
We should improve the commenting of these concepts. The commenting of sub_tile in https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/bb51a8b5b3ed65a8cb2ffebc6aa4b96c30f40630/libs/libarchfpga/src/physical_types.h#L668 should be improved.  The commenting of t_physical_type should similarly be improved around this concept.

t_logical_block should also have a clear distinction between logical and physical blocks.

#### Current Behaviour
I either can't find the data model comments on this, or they don't exist.

#### Possible Solution
We need both high-level data model comments around this, and clear commenting of all the types and data members involved.

acomodi, 2020-09-10:
@vaughnbetz FYI, opened this PR: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1531.

Please let me know if more details need to be added.

On a side note, there is more explanation on the sub tile and equivalent tiles in the tutorial section of the documentation that might provide more information:
- [Heterogeneous tiles](https://docs.verilogtorouting.org/en/latest/tutorials/arch/heterogeneous_tiles/)
- [Equivalent sites](https://docs.verilogtorouting.org/en/latest/tutorials/arch/equivalent_sites/)

vaughnbetz, 2025-05-19:
Closed as completed.

closed 2025-05-19 by vaughnbetz