verilog-to-routing/vtr-verilog-to-routing #1539: Warning from g++ in vtr_vector_map.h due to unsigned integer sign check

dpbaines, 2020-09-13:
#### Expected Behaviour

There should be no or very minimal warnings during the build process. This means that when there are warnings or errors we can see them clearly.

#### Current Behaviour

Updating g++ has created the warning output I posted below when building VPR. This warning is generated everytime the vector_map class is referenced, which is a lot. This means the output is completely spammed with warnings.

```
/home/dbaines/Documents/research/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_vector_map.h: In instantiation of ‘vtr::vector_map<K, V, Sentinel>::const_reference vtr::vector_map<K, V, Sentinel>::operator[](K) const [with K = vtr::StrongId<cluster_port_id_tag>; V = vtr::StrongId<cluster_block_id_tag>; Sentinel = vtr::DefaultSentinel<vtr::StrongId<cluster_block_id_tag> >; vtr::vector_map<K, V, Sentinel>::const_reference = const vtr::StrongId<cluster_block_id_tag>&]’:
/home/dbaines/Documents/research/vtr-verilog-to-routing/vpr/src/base/netlist.tpp:217:24:   required from ‘BlockId Netlist<BlockId, PortId, PinId, NetId>::port_block(PortId) const [with BlockId = vtr::StrongId<cluster_block_id_tag>; PortId = vtr::StrongId<cluster_port_id_tag>; PinId = vtr::StrongId<cluster_pin_id_tag>; NetId = vtr::StrongId<cluster_net_id_tag>]’
/home/dbaines/Documents/research/vtr-verilog-to-routing/vpr/src/base/netlist.tpp:327:12:   required from ‘BlockId Netlist<BlockId, PortId, PinId, NetId>::pin_block(PinId) const [with BlockId = vtr::StrongId<cluster_block_id_tag>; PortId = vtr::StrongId<cluster_port_id_tag>; PinId = vtr::StrongId<cluster_pin_id_tag>; NetId = vtr::StrongId<cluster_net_id_tag>]’
/home/dbaines/Documents/research/vtr-verilog-to-routing/vpr/src/base/netlist.tpp:394:16:   required from ‘BlockId Netlist<BlockId, PortId, PinId, NetId>::net_driver_block(NetId) const [with BlockId = vtr::StrongId<cluster_block_id_tag>; PortId = vtr::StrongId<cluster_port_id_tag>; PinId = vtr::StrongId<cluster_pin_id_tag>; NetId = vtr::StrongId<cluster_net_id_tag>]’
/home/dbaines/Documents/research/vtr-verilog-to-routing/vpr/src/timing/PostClusterDelayCalculator.tpp:262:95:   required from here
/home/dbaines/Documents/research/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_vector_map.h:65:35: warning: comparison of unsigned expression in ‘>= 0’ is always true [-Wtype-limits]
   65 |         VTR_ASSERT_SAFE_MSG(index >= 0 && index < vec_.size(), "Out-of-range index");
      |                             ~~~~~~^~~~
/home/dbaines/Documents/research/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_assert.h:103:34: note: in definition of macro ‘VTR_ASSERT_IMPL_NOP’
  103 |         static_cast<void>(sizeof(expr)); \
      |                                  ^~~~
/home/dbaines/Documents/research/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_vector_map.h:65:9: note: in expansion of macro ‘VTR_ASSERT_SAFE_MSG’
   65 |         VTR_ASSERT_SAFE_MSG(index >= 0 && index < vec_.size(), "Out-of-range index");
      |         ^~~~~~~~~~~~~~~~~~~
```

#### Possible Solution

Remove the index >= 0 check fixes this.

#### Steps to Reproduce

1. Build VPR with g++ 10.2.0

#### Context

#### Your Environment

* VTR revision used: VTR 8.0
* Operating System and version: Linux-5.8.5-arch1-1 x86_64
* Compiler version: GNU 10.2.0

jeanlego, 2020-09-22:
I am getting the same error.
@dpbaines where you gonna make a PR for it? I am more than happy to do so otherwise.

dpbaines, 2020-09-22:
I can make one right now, I was just waiting to make sure it wasn't just me!

jeanlego, 2020-09-22:
> I can make one right now, I was just waiting to make sure it wasn't just me!

I should've stepped up earlier I guess!

Thanks :)

closed 2020-09-29 by vaughnbetz