verilog-to-routing/vtr-verilog-to-routing #1546: Packer assertion failure on clock buffer primitive

mkurc-ant, 2020-09-16:
I want to make a net of a top-level input port become a clock source so that I can constrain it using SDC directive(s).

The target FPGA architecture requires that there is a `CLOCK_CELL` IO buffer primitive at a clock input that drives a global clock distribution buffer `GMUX_IP`. From there the clock can be routed to flip-flops. I want the clock to be propagated through both of these primitive cells.

I modeled both `CLOCK_CELL` and `GMUX_IP` primitives as clock buffers according to: https://docs.verilogtorouting.org/en/latest/tutorials/arch/timing_modeling/#clock-buffers-gates

When VPR is run it correctly identifies top-level inputs as clock sources but fails on an assertion during pre-packing.

#### Expected Behaviour
Packer succeeds, input ports `clk(0)` and `clk(1)` are considered as netlist clock sources.

#### Current Behaviour
Netlist clock sources are identified fine but packing fails with `vpr/src/pack/prepack.cpp:1127 get_driving_block: Assertion 'pin_number == 1' failed.` error.

#### Possible Solution
Unclear. Simply commenting out the assertion makes the packer pass (at least for the attached design).

The problem seems not to be related to "multi-clock primitives" as the TODO comment in the code suggests: `//TODO: support multi-clock primitives`. I'd like to know why the assertion is there and why `pin_number` has to be 1.

#### Steps to Reproduce
1. Use the architecture file and EBLIF netlist from the attached archive:  [pack.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5231842/pack.tar.gz)
2. Point the env. variable `VPR` to the VPR binary and run `runme.sh`

#### Context
I want to make a net of a top-level input port become a clock source so that I can constrain it using SDC directive(s).

#### Your Environment
* VTR revision used: 8.1.0-dev+7c167f8c8
* Operating System and version: Linux Ubuntu 18.04 LTS
* Compiler version: GNU 7.4.0 on Linux-4.15.0-45-generic x86_64

mkurc-ant, 2020-09-17:
@kmurray Could you take a look at it?

mkurc-ant, 2020-09-17:
@vaughnbetz Could you take a look at it?

vaughnbetz, 2020-09-22:
The proposed fix looks OK to me; see PR for comments.