verilog-to-routing/vtr-verilog-to-routing #1577: RRNode side method no longer correct

litghost, 2020-10-27:
#### Context

#1448 changed the rr graph build to only generate one IPIN/OPIN RR node even if the relevant cluster had pins on multiple sides of the tile.  However when this change was made, the underlying RRNode method for getting the sides of a node was not changed, see:

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/37095c742471af2e2c8d90faf10ab13156a47942/vpr/src/route/rr_graph_storage.h#L189-L195

`node_side` still returns just one value, even if the underlying RR node is actually representing multiple sides.

Code like: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/37095c742471af2e2c8d90faf10ab13156a47942/vpr/src/route/check_rr_graph.cpp#L565-L576 will stop working.

@tangxifan / @vaughnbetz 

#### Possible Solution

Now that IPIN/OPIN nodes can represent pins on multiple sides of the tile, the relevant node methods need to be updated to account for this.  Possible solutions:
 - Change `side()` and `node_side()` to `sides() and `node_sides()`
 - Remove `side()` and `node_side()` and add "is_on_side()`
 
Etc

tangxifan, 2020-10-28:
@litghost I noticed this issue when fixing the bug in routing resource graph. I have the same idea on changing the `side()` and `node_side()` method to `sides()` and `node_sides()`. But I am not sure what the `is_on_side()` will do? Does it record which side is used by the router?

litghost, 2020-10-28:
The signature of `is_on_side()` would be `bool RRNode::is_on_side(RRNode node, Side side)`, and returns true if the rr node connects to that side.  This would be a scalar version of `sides` or `node_sides`.

tangxifan, 2020-10-28:
@litghost Got it. Do you expect me to fix this on another pull request? Or you have a solution in the Symbiflow version?

litghost, 2020-10-28:
> @litghost Got it. Do you expect me to fix this on another pull request? Or you have a solution in the Symbiflow version?

We are have a workaround, but the current VPR data model in upstream is broken until this is fixed.  I'm surprised `check_rr_route` didn't fail somewhere, because we both agree that `node_side` no longer works correctly.  I suspect we've just gotten lucky so far.

vaughnbetz, 2020-10-28:
Yup, we should fix.  Depending on what the callers want, I am fine with coding only sides / node_sides, or also coding is_on_side if callers would also benefit from that.

closed 2021-02-02 by vaughnbetz