verilog-to-routing/vtr-verilog-to-routing #1607: Remove checking all subtile locations for placement

MohamedElgammal, 2020-12-11:
Longer term:  I don't think we really need usage anymore (really need to check all the sub_tile_locations to see what's there).  So I think it could be deleted from the code base (but haven't looked really carefully).  Suggest a separate issue to track that.

_Originally posted by @vaughnbetz in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1570#discussion_r540375018_

vaughnbetz, 2025-05-25:
We still have a .usage method on the grid, and an associated t_grid_blocks .usage data member. We have set_usage, get_usage, increment_usage and decrement_usage functions, but nothing seems to actually use this data on the grid. We should delete it ... we no longer care how many blocks are placed at a certain tile; we iterate through all possible sub_tiles at a grid location to see what has been placed there.
@haydar-c can you delete these routines and data member, verify all is well, and make a PR?

haydar-c, 2025-05-26:
Sure, I’ll be looking into it.