verilog-to-routing/vtr-verilog-to-routing #1616: Differentiate clock from other routing (better filtering in graphics)

vaughnbetz, 2020-12-17:
Tuo Xie requested the ability to visualize clocking (pins & wires I believe). We should make it easier to visualize subsets of the device routing resource and types of routing nets.

#### Proposed Behaviour
Add filtering to the rr_nodes displayed (ToggleRR), and to the nets displayed (ToggleNets). I think we should have an option to filter what is shown by node type, node name (segment or pin type name with wildcarding), the metadata symbiflow can add, and possibly other features.  A key use case would be to see if clock networks could be shown separately from other routing, for architectures that used a suitable arch.xml and naming of wires and pins in that file.

Similarly we should allow some additional filtering of nets in ToggleNets: show nets with certain names or that drive pins of the clock type.

#### Current Behaviour
We have pretty limited abilities to filter the drawing of rr-nodes or nets. Documentation on that is at https://docs.verilogtorouting.org/en/latest/vpr/graphics/ . This issue is to give us more general filtering capabilties.

xtthu, 2020-12-18:
Besides the filtering function added to the rr_nodes displayed (ToggleRR) and the nets displayed (ToggleNets), I suggest to take consideration of some special cases (such as the element <clocknetworks> defined in the xml file, which specifies the clock architecture). If this clock network distribution (i. e. the H-tree) could be visualized in the VPR’s graphics conveniently, it would be helpful to view the clock driving status as well as implement timing analysis.

Best regards,
Tuo Xie

vaughnbetz, 2020-12-18:
Thank Tuo.  That's a good idea; wires under the <clock_network> tag could have some additional information (a flag, or some metadata) that identifies them as part of the clocking network, and we could make that a setting.

vaughnbetz, 2025-05-25:
Re-opening this and sending to @SamuelHo10 to consider as he updates the graphics.