verilog-to-routing/vtr-verilog-to-routing #1617: De-clutter vpr clustering output (and some other output ideas)

vaughnbetz, 2020-12-17:
- vpr prints too much information in clustering. The detailed per cluster information should be deleted or printed only at high verbosity (or to an echo file). Instead we should indicate progress (% of design clustered, with updates every 10% or so and a cpu time so far and a few key stats).  Once clustering completes we should print some key statistical outputs.

- we should also echo the command line arguments before we print architecture file or circuit statistics (i.e. echo back the options vpr was invoked with as early as possible, right after we echo back the architecture and circuit file names if possible).

sfkhalid, 2021-01-14:
-RoutingArch. lines should be either deleted or moved to be with the arch file description (probably better to be deleted)
-Printing out of all the CAD control parameters (the Opts) should come before we print the circuits stats, arch file

vaughnbetz, 2025-05-25:
This was completed; marking as such.