verilog-to-routing/vtr-verilog-to-routing #1625: Using rr graphs from earlier VPR version

mkurc-ant, 2021-01-08:
I'm trying to use a rr graph generated using the VPR version currently used in OpenFPGA project (can't establish exact VPR revision, the OpenFPGA revision is `a9f91513`) with the upstream VPR. I'm encountering the following assertion failure during post-load rr graph check:
```
Type: Routing
File: vpr/src/route/rr_graph2.cpp
Line: 1333
Message: Mismatch between RR node length (1) and count within rr_node_indices (1, should be length + 1): RR node: 120985 type: CHANY location: (32,31) <-> (32,32) track: 47 len: 1 longline: 0 seg_type: L4 dir: DEC_DIR capacity: 1 fan-in: 4 fan-out: 11
s.
```
When I commented out the assertion I managed to successfuly load and use the graph throughout the whole VPR flow. 

#### Expected Behaviour
The graph should load without assertion errors.

#### Current Behaviour
An assertion is triggered which makes VPR crash.

#### Possible Solution
Know what is causing the assertion violation and modify the graph somehow so its structure remains intact while it can be used by the upstream VPR.

#### Steps to Reproduce
1. Use the VPR from OpenFPGA to generate and write rr graph for a given arch.xml.
2. Use the same arch.xml plus the previously written rr graph with VPR from the current master branch.

#### Context
I want to use FPGA architectures from the OpenFPGA project with SymbiFlow. The important thing is to have consistent routing so I have to use identical rr graphs. The only way to ensure that is to use a rr graph generated by VPR from OpenFPGA with VPR used in SymbiFlow.

#### Your Environment

* VTR revision used:  8.1.0-dev+93af9f19d
* Operating System and version: Linux, Ubuntu 18.04
* Compiler version: GNU 7.4.0 on Linux-4.15.0-45-generic x86_64

mkurc-ant, 2021-01-08:
@tangxifan Could you please take look at this issue? Is it relevant to what are you currently working on? (single node for multiple pin sides)

mkurc-ant, 2021-02-25:
I've discovered that the graph I'm trying to read has two `CHANY` nodes at the same location with identical `ptc` numbers:

```
<node id="153753" type="CHANY" direction="DEC_DIR" capacity="1">
	<loc xlow="32" ylow="31" xhigh="32" yhigh="32" ptc="47"/>
	<timing R="0" C="0"/>
	<segment segment_id="2"/>
</node>
<node id="153777" type="CHANY" direction="DEC_DIR" capacity="1">
	<loc xlow="32" ylow="32" xhigh="32" yhigh="32" ptc="47"/>
	<timing R="0" C="0"/>
	<segment segment_id="2"/>
</node>
```

This makes VPR overwrite the 153753 node with 153777 in the `t_rr_node_indices` array of the device context.

mkurc-ant, 2021-02-25:
I'm closing this issue as the problem I've encountered was not caused by any issue with VPR itself.

Meanwhile it may be worth adding a check if during rr graph loading nodes are not getting overwritten there: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/7be7cbe3d726da76fd6280e77fb5731fd4ab4873/vpr/src/route/rr_graph_uxsdcxx_serializer.h#L1638-L1666

closed 2021-02-25 by mkurc-ant

tangxifan, 2021-02-25:
@mkurc-ant  actually, this is due to staggered wires that are frequently seen in tileable routing architecture. 
To model staggered wires, a rr_node may have multiple ``ptc_num``.
However, VPR RRGraph only supports 1 ``ptc_num``. That is why you see duplicated ``ptc_num`` because the ``ptc_num`` given for each ``rr_node`` is the 1st one in the ``ptc_num`` list.