verilog-to-routing/vtr-verilog-to-routing #1632: Post-routing crash (SIGABORT) seen with some specific arch and design files

aman26kbm, 2021-01-15:
<This issue was reported on commit: https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/ea7acf1582ece35e892c26b756aa302d2e12ddb2.  Please see the comments there for historical details>

#### Expected Behaviour
There shouldn't be a crash.

#### Current Behaviour
There is a SIGABORT failure post-routing. Seeing this failure when the arch file has a local crossbar in the DSP block. Here's the last part of the stack trace:
#6 0x0000555558a05f3c in vtr::flat_map2<int, t_pb_route, std::less, std::vector<std::pair<int, t_pb_route>, std::allocator<std::pair<int, t_pb_route> > > >::operator[](int const&) const (this=0x60c0000c55f0, key=@0x7fffffff78b0: 118) at /export/aman/vtr_aman/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_flat_map.h:374
#7 0x000055555973998e in find_pb_route_clb_input_net_pin(vtr::StrongId<cluster_block_id_tag, int, -1>, int) (clb=..., sink_pb_pin_id=1694)
at /export/aman/vtr_aman/vtr-verilog-to-routing/vpr/src/util/vpr_utils.cpp:595
#8 0x00005555589efb1b in PostClusterDelayCalculator::atom_net_delay(tatum::TimingGraph const&, tatum::util::StrongId<tatum::edge_id_tag, int, -1>, DelayType) const (this=0x6120003a6b50, tg=..., edge_id=..., delay_type=DelayType::MAX) at /export/aman/vtr_aman/vtr-verilog-to-routing/vpr/src/timing/PostClusterDelayCalculator.tpp:258

#### Possible Solution

#### Steps to Reproduce

I was adding a new arch file to VTR. I saw this failure initially when I used the design temp.v and the arch agilex_arch.xml in this tar ball: https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5756158/vtr_crash.tar.gz

I did some binary searching and identified that this commit was causing the error: https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/ea7acf1582ece35e892c26b756aa302d2e12ddb2

After that I've debugged more and saw that the issue could be seen with spree.v as well (which is in the benchmarks/verilog directory in the repo). I removed features in the arch one by one and narrowed it down to see that failure only shows up when the DSP block has a local crossbar. That is, arch file agilex_arch.extra_modes.no_local_xbar.xml passes, but agilex_arch.xml fails. These arch files are in: https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5789978/arch_files.zip.

**So, to reproduce, there are two options:
1. Run the VTR flow with spree.v in benchmarks/verilog and agilex_arch.xml from https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5789978/arch_files.zip
2. Run the VTR flow with temp.v in https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5756158/vtr_crash.tar.gz and agilex_arch.xml from https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5789978/arch_files.zip**

#### Context

Trying to add a new arch file to VTR. As a workaround, I've removed the local crossbar in the DSP block in the architecture.

#### Your Environment

* VTR revision used: Master
* Operating System and version: Ubuntu 18.04
* Compiler version:

tangxifan, 2021-01-28:
@aman26kb Do you have the ``blif`` version of the ``spree.v`` benchmark? I can try to run it locally. What are the VPR options you were using?

aman26kbm, 2021-01-29:
I was running the standard VTR flow. So, the VPR options that flow uses were being used. I wasn't running VPR separately. If you run the VTR flow (using run_vtr_task.py), the blif file will get generated in the output directory.