verilog-to-routing/vtr-verilog-to-routing #1638: run genfasm unsuccessfully with error

yongjiangwang, 2021-01-21:
follow the quickstart manual to run blink through vtr flow and it seems I can get synthesis, place and route successfully with 8.0.0. version of download. But when I try to use genfasm in utils directory, I got the error 1 and program stopped.
#### Expected Behaviour

expect the program to generate blink.fasm
#### Current Behaviour

Error  1:
Type: Other
File:  fasm.cpp
Message: Failed to find LUT output definition.
#### Possible Solution

#### Steps to Reproduce

1. genfasm EArch.xml blink.blif --pack --place --route --route_chan_width 100 
2. 

#### Context

#### Your Environment

Ubuntu VirtualBox
* VTR revision used: vtr-verilog-to-routing-8.0.0
* Operating System and version:
* Compiler version:

balivila, 2021-02-22:
same problem
have you solved the problem?

yongjiangwang, 2021-02-22:
Actually the tool is not for this. After studying more on the tools, actually the tool is really for given RTL, logic cell arch, routine resoucese, how many logic blocks are needed to implement the given RTL. For me, I actually want to have a tool with the fixed x y boundaries, run synth, routing and generate bitstream.

yongjiangwang, 2021-02-22:
it can be considered as closed.

closed 2021-02-22 by yongjiangwang