verilog-to-routing/vtr-verilog-to-routing #1643: Generate statement seems to be resulting in code with multiple fanins for a net

aman26kbm, 2021-01-27:
Attached zip file has a piece of Verilog code that instantiates 8 hard blocks using a generate statement. The hard blocks happen to be RAMs here but that shouldn't matter.

Each hard block's output signal is 32 bits and separate and are muxed in the end. Also in the attached zip is a schematic from Vivado to show the structure of the design.

Looking at ODIN's generated output (blif), I see the same net connected to the output ports of multiple hard block instances. That results in an error from ABC (next stage in VPR): 
abcFanio.c:93: Abc_ObjAddFanin: Assertion `!Abc_ObjIsNet(pObj) || !Abc_ObjFaninNum(pObj)' failed.

The error doesn't specify the exact line or the name of the signals it thinks is causing the failure. But my debug says that it is because of the outputs being ganged together.

The arch used was: k6_frac_N10_frac_chain_depop50_mem32K_40nm.xml

When I replace the code with 8 manual instances, I don't see the error. That code is in a commented out section in the Verilog file.

[generate_issue.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/5882760/generate_issue.zip)

sdamghan, 2021-02-02:
That's really odd! Thanks for mentioning @aman26kbm 
@nanjekyejoannah would you please have a look?

aman26kbm, 2021-02-21:
Can you please update the assignee field, @sdamghan ?

nanjekyejoannah, 2021-02-22:
Yes, working on this...please assign.

aman26kbm, 2021-03-05:
Hi @nanjekyejoannah , just wanted to check if this has been fixed. Please let me know. Thanks.

nanjekyejoannah, 2021-03-05:
Am opening a PR tomorrow, after some verification, sorry I had some school work holding me this week.

nanjekyejoannah, 2021-03-08:
This is my analysis, the way you are doing block instantiation in the generate statement seems strange.
See the use of iter, compare to how you did the same in the manual instances.
```
genvar iter;
generate
	for (iter=0; iter<8; iter=iter+1) begin: U
		dual_port_ram u_dual_port_ram(
        .addr1(low_addr_a),
        .we1(actual_wren_a[iter]),
        .data1(data_a),
        .out1(q_a_long[32*iter+:32]),
        .addr2(low_addr_b),
        .we2(actual_wren_b[iter]),
        .data2(data_b),
        .out2(q_b_long[32*iter+:32]),
        .clk(clk)
        );
	end
endgenerate
```
For example, the first instance from the loop will be:

```
dual_port_ram u_dual_port_ram_0(
           .addr1(low_addr_a),
           .we1(actual_wren_a[0]),
           .data1(data_a),
           .out1(q_a_long[0:32]),
           .addr2(low_addr_b),
           .we2(actual_wren_b[0]),
           .data2(data_b),
           .out2(q_b_long[0:32]),
           .clk(clk)
           );
```
    

Instead of the intended:

```
dual_port_ram u_dual_port_ram_0(
           .addr1(low_addr_a),
           .we1(actual_wren_a[0]),
           .data1(data_a),
           .out1(q_a_long[31:0]),
           .addr2(low_addr_b),
           .we2(actual_wren_b[0]),
           .data2(data_b),
           .out2(q_b_long[31:0]),
           .clk(clk)
           );
    
```
If you want to test, copy one of the manual instances to the generate statement, no loop, and the output files will match.
Please let me know if am missing something. 

I tried to make changes but it looks like I was changing the output files to match the behavior in your current generate statement.

nanjekyejoannah, 2021-03-08:
This is my analysis, the way you are doing block instantiation in the generate statement seems strange.
See the use of iter, compare to how you did the same in the manual instances.
```
genvar iter;
generate
  for (iter=0; iter<8; iter=iter+1) begin: U
    dual_port_ram u_dual_port_ram(
        .addr1(low_addr_a),
        .we1(actual_wren_a[iter]),
        .data1(data_a),
        .out1(q_a_long[32*iter+:32]),
        .addr2(low_addr_b),
        .we2(actual_wren_b[iter]),
        .data2(data_b),
        .out2(q_b_long[32*iter+:32]),
        .clk(clk)
        );
  end
endgenerate
```
For example, the first instance from the loop will be:

```
dual_port_ram u_dual_port_ram_0(
           .addr1(low_addr_a),
           .we1(actual_wren_a[0]),
           .data1(data_a),
           .out1(q_a_long[0:32]),
           .addr2(low_addr_b),
           .we2(actual_wren_b[0]),
           .data2(data_b),
           .out2(q_b_long[0:32]),
           .clk(clk)
           );
```

Instead of the intended:

```
dual_port_ram u_dual_port_ram_0(
           .addr1(low_addr_a),
           .we1(actual_wren_a[0]),
           .data1(data_a),
           .out1(q_a_long[31:0]),
           .addr2(low_addr_b),
           .we2(actual_wren_b[0]),
           .data2(data_b),
           .out2(q_b_long[31:0]),
           .clk(clk)
           );

```
If you want to test, copy one of the manual instances to the generate statement, no loop, and the output files will match.
Please let me know if am missing something.

I tried to make changes but it looks like I was changing the output files to match the behavior in your current generate statement.

Of course, the generate statement as is runs without errors but other steps like ABC will complain about the output files produced.

aman26kbm, 2021-03-08:
Thanks, @nanjekyejoannah 

Can you please provide the code for the generate loop that I should use?

I think 
`q_a_long[32*iter+:32]`
should expand to:
```
q_a_long[31:0]
q_a_long[63:32]

```And so on, when iter increases from 0 to 1 to 2...

sdamghan, 2021-03-08:
@aman26kbm I think this code works, however, would you please test it and let us know about the final results?
```
genvar iter;
generate
	for (iter=0; iter<2; iter=iter+1) begin: U
		dual_port_ram u_dual_port_ram(
        .addr1(low_addr_a),
        .we1(actual_wren_a[iter]),
        .data1(data_a),
        .out1(q_a_long[32*iter+:32*iter]),
        .addr2(low_addr_b),
        .we2(actual_wren_b[iter]),
        .data2(data_b),
        .out2(q_b_long[32*iter+:32*iter]),
        .clk(clk)
        );
	end
endgenerate
```

aman26kbm, 2021-03-08:
I can try that, but this
`q_a_long[32*iter+:32*iter]`
is not allowed in Verilog. The width expression (the part after the `+:`) should be a constant.

nanjekyejoannah, 2021-03-08:
Apologies, I may have missed something.

Please test and confirm. I think the generate statement itself has no issue..its maybe how we are dealing with instantiating the blocks.

sdamghan, 2021-03-08:
@aman26kbm  Yes, that's true. You are right.
@nanjekyejoannah  I have just tried the following codes in Odin, the simulation results of both were the same. I have also attached the input and output vector for the simulation. Would you please have a look at blif files? Moreover, Odin does not throw an error while facing with non-constant value after ``+:`` operation however it should.

[generate_statement.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6104502/generate_statement.zip)

nanjekyejoannah, 2021-03-08:
@sdamghan , so to see if I understand, are we now solving to make sure Odin throws the error for the faulty behaviour of using non-constant values when instantiating the blocks? Then this means we are now investigating another issue related to how Odin 
is handling instantiating.

sdamghan, 2021-03-08:
@nanjekyejoannah first, please make sure the provided code working properly (i.e. generate loop instantiates the hard blocks correctly and produces the same results as we instantiate them manually and outside of the loop). Moreover, please see why Odin does not show any error while facing a non-constant (the value of ast_node is not vNUMBER).

nanjekyejoannah, 2021-03-08:
@sdamghan, I think 

1. the fact that we can not have non-constants is why instantiating hard blocks this way i.e  in a loop is problematic in the first place but let us talk about this tomorrow I guess.
2. I think I can go on to work on the check to show an error for constants in this scenario as we think about 1

sdamghan, 2021-03-08:
>     1. the fact that we can not have non-constants is why instantiating hard blocks this way i.e  in a loop is problematic in the first place but let us talk about this tomorrow I guess.

@nanjekyejoannah Please consider this as well, http://staff.ustc.edu.cn/~songch/download/IEEE.1364-2005.pdf#G6.118955

sdamghan, 2021-05-23:
@nanjekyejoannah Would you please provide an update on this issue? I would highly recommend you to create a PR and push your changes

nanjekyejoannah, 2021-06-03:
Please unassign me from this issue. Its clear I wont get back to it.

I had to context switch to the project, at the point where we thought, ODIN was not reporting errors on non-constants, which I confirmed was false.

For anyone taking on the issue, I will leave the following pointers as shared by @sdamghan.

As a clue, please consider the following:

- Decrease the the number of for iteration to 2 (as well as manual instantiation)
- Have a look at the process that generate for keyword would pass in the Odin.
- We definitely have a step that shrink the generate for into the number of iteration.
-  Afterwards, please find out how these shrinked nodes will be proceed in netlist creation.
- At this point, Odin probably connects the output of each mem instance to the same block.
- Here you need to create separate nets for each block as the same as manual instantiation.
  
  In terms of memory block, they are pretty much like a black box with 3+1/6+1 input ports and 2 output ports as follow:

```
  input clock; // 1 bit
  input addr1/addr2; // m bits
  input data1/data2; // n bits
  input writeEnable1, writeEnable2; // 1 bit
   
  output out1/out2; // n bits
```