verilog-to-routing/vtr-verilog-to-routing #1644: Support for $signed missing in ODIN

aman26kbm, 2021-01-27:
Currently ODIN reports an error when an occurrence of $signed is seen in the HDL file. Requesting support for $signed through this ticket.

#### Expected Behaviour
No error should be thrown. The argument to $signed should be treated as a signed number. I don't think simple arithmetic operations (like addition or subtraction) will need to change. Main thing to update is how right shifts and comparisons are handled. But there may be more things required to fully support $signed.

#### Current Behaviour
ODIN reports an error when an occurrence of $signed is seen in the HDL file.
error[NETLIST]:  Operation not supported by Odin

#### Possible Solution

#### Steps to Reproduce
I don't think a testbench is required here. It's just about $signed operator/function. Since $signed can be used anywhere, a simple testbench may not be enough.

Here is the code that was in the file I was working with when I came across the error:

reg  main_32_33;
reg [31:0] main_fusion1loop_bodydim1_22_reg;
main_32_33 = ($signed(main_fusion1loop_bodydim1_22_reg) > $signed(32'd1118925335));

So to start with, I'd say a simple verilog file with code like the above should cause the error to be reproduced..

#### Context

#### Your Environment

* VTR revision used: Master
* Operating System and version: Ubuntu 18.03
* Compiler version:

aman26kbm, 2021-02-21:
Adding @sdamghan to assign

sdamghan, 2021-02-22:
Thanks @aman26kbm , The signed support has been already underworking. To solve the problem now, I recommend you to put the value of a signed value manually into a variable.
I'll also update you on this topic in the future :)

closed 2025-05-25 by github-actions[bot]