verilog-to-routing/vtr-verilog-to-routing #1648: vtr weekly runs not completing

vaughnbetz, 2021-01-28:
The vtr weekly runs are not completing.  We should take a look at why.

#### Expected Behaviour
These should complete in kokoro

#### Current Behaviour
It almost always fails, due to some jobs not finishing.  Keith looked at it and it is limited to 3 cores in kokoro. We are running on a 60 GB machine, so we can't push the core count very high.  Kevin ordered the benchmarks in the script so high memory circuits are less likely to run at the same time.

#### Possible Solution
Log file parsing is currently slow:  wait for https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1647 to be resolved before analyzing more.
Shorten weekly run, or assign more resources in kokoro.  We could push the parallelism higher (j = 6 might be reasonable) if we move to a bigger machine.
Gaussian_blur is the slowest design; perhaps not running it or running it in a separate CI test would help.

#### Steps to Reproduce
Unfortunately, only google can see what is happening on kokoro when it is running.  I suggest running locally and seeing how much CPU time we're using.

#### Context
We're not getting weekly regression test coverage.  It's not as key as nightly, but would be useful as it includes the high effort Titan.

#### Your Environment
Kokoro CI

mithro, 2022-02-24:
Is this issue still relevant with the move to GitHub Actions?

mithro, 2022-02-24:
Given that it was from early 2021 and we are now in 2022, I'm going to close it.

closed 2022-02-24 by mithro