verilog-to-routing/vtr-verilog-to-routing #1655: Assertion 'parent_pb->mode == pb_graph_node->pb_type->parent_mode->index' failure

mkurc-ant, 2021-02-03:
Recently I've encountered the assertion failure during clustering: `vpr/src/pack/cluster.cpp:1437 try_place_atom_block_rec: Assertion 'parent_pb->mode == pb_graph_node->pb_type->parent_mode->index' failed.`

I've spent some time debugging the code and I've roughly narrowed the problem down. For some reason `child_pbs` together with a selected mode of a `pb`  are not cleared correctly. Those are allocated and set as packer tries to place an atom. Every time a placement fails those should be cleared but in rare cases they are not.

The same assertion gets triggered in some cases for architectures with disabled modes (with `disable_packing = "true"`). The packer allocates `pb` and its `child_pbs` for such modes and bails out later.

#### Expected Behaviour
Packer should either succeed or fail but without triggering assertions.

#### Current Behaviour
The assertion 'parent_pb->mode == pb_graph_node->pb_type->parent_mode->index'  at ``vpr/src/pack/cluster.cpp:1437 try_place_atom_block_rec` gets triggered.

#### Possible Solution
Add (or fix?) cleanup code that removes `child_pbs` created during atom placement if it fails for any reason.

#### Steps to Reproduce

Not yet.

#### Context

I've been working for supporting a VPR architecture that is used in the OpenFPGA project in SymbiFlow (https://github.com/lnis-uofu/SOFA/blob/master/ARCH/vpr_arch/k4_N8_tileable_reset_softadder_register_scan_chain_nonLR_caravel_io_skywater130nm.xml)

#### Your Environment

* VTR revision used: GIT `fa683da07`
* Operating System and version: Linux Ubuntu 18.04
* Compiler version: GNU 7.4.0 on Linux-4.15.0-45-generic x86_64

tangxifan, 2021-02-04:
If it is related to the ``disable_in_packing``, I am the author of that part. I am glad to help on this issue

mkurc-ant, 2021-02-04:
@tangxifan It is related to that but not only to. I've been able to trigger the issue even with no disabled modes in the arch. However, the bug is pretty hard to trigger. It showed up in a couple of QuickLogic test designs. 

If you run VPR with the architecture and circuit I've attached (`design.tar.gz`) you will see that it fails during packing a `.latch` into the `io` pb_type (there are registered IOs in the arch). This is the "k4_N8" arch from SOFA processed to be used with SymbiFlow.

tangxifan, 2021-02-05:
@mkurc-ant No worries. I managed to reproduce this error in other OpenFPGA architectures.

closed 2021-02-25 by vaughnbetz