verilog-to-routing/vtr-verilog-to-routing #1665: Support arrays in ODIN

aman26kbm, 2021-02-16:
Currently, ODIN doesn't support any usage of arrays other than as memories. But there are many places where arrays are useful, merely as a way to make coding easy and methodical (instead of handling 1D bit-vectors). Here are some examples:

wire [7:0] arr[3:0];
assign arr[3] = 8'h34;
assign arr_added = arr[0] + arr[1] + arr[2] + arr[3];
for (i=0; i<3; i=i+1)
  if(arr[i] = arr[i+1]) <do_something>

This is just one dimension. Sometimes, multiple dimensions are useful too: `wire [7:0] arr[3:0][4:0];` But this is lower priority.

Adding support for arrays in ODIN will be very helpful by enabling many designs to use the VTR flow.

aman26kbm, 2021-02-21:
Adding @sdamghan to assign

rourabpaul1986, 2021-03-06:
I have also faced the same issue in Odin II, multi dimensional reg array is supported but multi dimensional wire array is not supported here

aman26kbm, 2021-04-11:
Moving bit slicing support problem from email to this github issue for better tracking:

[Aatman]
Example code:
```
module test (
 output [32-1:0] out
);

reg [63:0] src1 [4:0];
assign out = src1[0][31:0];

endmodule
```

This gives an error, even on the branch where array support is enabled.

[Seyed]
   Actually, the aforementioned syntax is not yet supported for bit slicing in arrays (Some array problems like two brackets usage (2D or similar) have still remained). However, the infrastructure for processing it is complete. I'll have a look at this and let you know whenever I update the branch.
 
    For now, if would like to assign the least 32 bits of an array element to a signal you can use the following syntax:

   assign out = src1[0];

    It will automatically assign the least 32 (i.e. the size of out signal) bits of src1[0] to the out signal.

closed 2025-05-25 by github-actions[bot]