verilog-to-routing/vtr-verilog-to-routing #1666: vtr nightly runs not completing

sfkhalid, 2021-02-17:
The vtr nightly runs are not completing. They may not be running at all in some cases. For example, the nightly run in this pull request #1641 says that the tool failed and duration is 00:00. The same thing is happening after trying to run nightly a few times on this pull request. The same issue also seems to be occurring on other pull requests - #1663 and #1658 are some examples of this. 
Also, it is not possible to look at the invocation log to see what is happening because it says "Unable to retrieve invocation log"

#### Expected Behaviour
The nightly run should have an elapsed runtime and either pass or fail. If the nightly test fails, some reasons for the failure should be able to be seen in the invocation log (ex. to show if some QoR tests failed).

#### Current Behaviour
The nightly test gets stuck on the yellow status and after some time just shows failed and duration 00:00. All other checks seem to be working fine.

#### Possible Solution
Not sure of what might resolve this. First step is probably to run nightly test on a pull request and see what is happening on kokoro when it is running.

#### Steps to Reproduce
Run nightly test on a pull request.

#### Context
Haven't been able to merge code on a pull request because nightly is not passing.

#### Your Environment
Kokoro CI

#### Additional Information
When I run the nightly regression test on my local machine, I get the following error:
[image]

I ran make get_titan_benchmarks and make get_ispd_benchmarks to make sure I had the proper benchmarks to run nightly so the issue is not from that. Also, other reg tests such as basic and strong run fine.

mithro, 2021-02-17:
FYI - @litghost

sfkhalid, 2021-02-18:
Solution: clean up log files on CI and run nightly again
Link with instructions: 
https://github.com/SymbiFlow/symbiflow-arch-defs/blob/45c83a2592923b9a785aa05c9464b5f453ebd151/.github/kokoro/package_results.sh#L43-L48

sfkhalid, 2021-02-22:
@litghost There seems to be an issue with vtr_reg_coffe having QoR failures that is contributing to the nightly test failures.

I first saw these failures in this pull request #1637. I thought they were odd since in this pull request, I only changed the order of how things are printed in the vpr log file.

Then, I ran nightly on my other branch, (in a pull request here #1641) which has significantly more changes than the other branch. All the tests passed except for vtr_reg_coffe, and again the same values failed QoR - pack_time, place_time, min_chan_width_route_time, and crit_path_route_time - as shown below. These four values are shown as being very small. Here are the failure messages:
[image]

I looked on some other pull requests to see if any other branches are having this test fail, and by looking at the log file, I found that #1667 had the same coffe QoR failures in the nightly run.

Finally, I ran the nightly test on master (I made sure that I had the latest version) and found that it had the same QoR failures as well, as shown below.
[image]

Given that the four failing values all have to do with time, it seems that for this architecture, the time is not being measured correctly somehow, since none of the other values seem to be affected. I am trying to look into why this may be happening, but wanted to address this issue here as well.

I also tried looking at some of the recent pull requests that have passed all tests, such as #1649, to see if the coffe tests had passed on them, but it seems they are not running nightly and that's why they passed.

vaughnbetz, 2021-02-25:
The golden results for the vtr_reg_coffe look wrong. The pack times, place times and channel widths on these golden results are much higher than what we are achieving, so we're failing the tests by having too good a result.  I think the golden results need to be updated, or there is some other issue. @aman26kbm can you take a look?

That's not the fundamental issue for the vtr nightly runs not completing, but it should be fixed.

aman26kbm, 2021-02-26:
Sure I can take a look. But I won't be able to get to this anytime soon. I can try to get to it late next week. Is that okay? If this is urgent, please let me know.

litghost, 2021-02-26:
I've identified one of the problems and fixed it, see #1674.  @acomodi had already added some logic to cleanup the workspace to drop the working space below 100G.  As you can see from https://source.cloud.google.com/results/invocations/b7799e26-286f-4691-820a-3193adb4cd3b/log , the working space is at 89 GB  with cleanup, not the 25 GB that @sfkhalid on their run.  Now that the nightly is finishing, @sfkhalid it would be a good plan to identify where the 89 GB is coming from.  If the packer log fixes dramatically lowers the disk size that would be great, but I don't believe it is going to get from here to their.

Here is the output master on the working space size:
```
+++ du -sh
++ echo 'Working directory size: 89G	.'
Working directory size: 89G	.
```

Here is where the working directory size is being reported: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/21ec4333283249c0017e071f6402228191864d88/.github/kokoro/steps/vtr-test.sh#L71-L76

@sfkhalid I recommend adding `du -h --max-depth 2` to get better visibility into where in the working directory disk space is going.

litghost, 2021-02-26:
To be clear, because the nightly working space directory is at 89 GB, it is fairly close to exceeding the max.  While right now, nightly runs are completing, it is easy to expect that to change in the future.  I'm leaving this issue open to see if we can trim the nightly working directory some more.

sfkhalid, 2021-02-26:
I ran the du command as suggested and here are some findings:

First to summarize: I ran nightly three times on my local machine, once it did not run fully because I had not made the symbiflow benchmarks and the next two times it ran through fully. The commands I used to run nightly were as follows:
make get_symbiflow_benchmarks
make get_titan_benchmark
make get_ispd_benchmarks
./run_reg_test.py vtr_reg_nightly -j3

When I ran du -h --max-depth 2 on the root directory, I found that out of the 25 GB total, 23 GB were coming from vtr_flow. All other directories had sizes less than 1 GB. Within vtr_flow the majority of the space (22 GB) was taken up by tasks/regression_tests/.

Within the regression_tests directory, a big chunk of space was taken up by the titan_quick_qor test in vtr_reg_nightly
[image]
 
The titan_quick_qor directory gave the following breakdown
[image]

From here I went into run002/stratixiv_arch.timing.xml/
[image]

I looked at the directory for the bitcoin miner circuit and noticed that although the directory size was said to be 598 MB, there were two files (the .net files) in the folder that were 2 GB each. 

[image]

I am not sure what is causing the discrepancy here - the directory is 598 MB according to du -sh, but when I do ls -l I can see that the .net files are 2 GB each. This may be why I am seeing a lower value like 25 GB for the total directory. Also, it seems that the .net files may be something we have to trim down if we want to reduce the nightly working directory more.

litghost, 2021-02-26:
> I ran the du command as suggested and here are some findings:

I was suggesting adding the "du" commands to the CI script so you can get a report from the CI machine.

sfkhalid, 2021-02-26:
> > I ran the du command as suggested and here are some findings:
> 
> I was suggesting adding the "du" commands to the CI script so you can get a report from the CI machine.

Oh I see, by CI script do you mean vtr-test.sh?

litghost, 2021-02-26:
> 
> 
> > > I ran the du command as suggested and here are some findings:
> > 
> > 
> > I was suggesting adding the "du" commands to the CI script so you can get a report from the CI machine.
> 
> Oh I see, by CI script do you mean vtr-test.sh?

Ya, `vtr-test.sh`.

acomodi, 2021-03-02:
FYI, as @litghost pointed out I had added some logic to clean the benchmarks data, but that did not prove to be enough.

I remember noticing that the circuits files from the "big" tests such as the Titan ones were copied in the run directory instead of using a symlink or such. 
I believe this is a good route to pursue to reduce disk-usage, as well as possibly allowing the increase of the number of cores used in the Nightly run ([continuous](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/4acad0fb53a45b4f2b6c1b15aafaf72fef88af85/.github/kokoro/continuous/nightly.cfg#L58) and [presubmit](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/4acad0fb53a45b4f2b6c1b15aafaf72fef88af85/.github/kokoro/presubmit/nightly.cfg#L58)), as the copy of these files might require quite some RAM, which ended up in a CI failure due to some memory exhaustion.

sfkhalid, 2021-03-03:
I add the suggested command du -h --max-depth=2 to the CI script and the results show that out of the 89 G, 87 G are coming from vtr_flow/tasks. 
[image]

The full log file is below which has the full breakdown of space.
[kokoro-sisyphus_resultstore_prod_foss-fpga-tools_verilog-to-routing_upstream_presubmit_nightly_dc001930-37a6-48c4-8540-5ff7a675a0e3_build.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6076656/kokoro-sisyphus_resultstore_prod_foss-fpga-tools_verilog-to-routing_upstream_presubmit_nightly_dc001930-37a6-48c4-8540-5ff7a675a0e3_build.log)

vaughnbetz, 2021-03-03:
Thanks.  Sounds like we need to start cleaning up some files, starting with the .net files.  
1. Sarah, can you add a little more detail CI script du command?  You can explore interactively on your local version to see what works and gives a reasonable amount of info (useful, but not overwhelming).  I think adding something like:
cd vtr_flow/tasks
du -h --max_depth=2   // or maybe max_depth = 3?

Would also be useful to identify which tasks are taking the most disk space.

2. We should change CI to reduce disk space.  Removing the .net files by default (not sure where to best run this, maybe in run_vtr_task) would be a good place to start.  sym_links instead of copies for big netlist files may also be useful, as Alessandro suggests, but I think removing the .net files is the biggest bang.

sfkhalid, 2021-03-03:
Yes, I was just looking into that. I'll run it now so nightly can finish running by the symbiflow meeting tomorrow.

sfkhalid, 2021-03-04:
From running with a greater -max-depth in the CI script, most of the space is coming from vtr_reg_nightly, and from vtr_reg_nightly a large amount of space is taken up by the titan_quick_qor test.
[image]
 The full log file is here : 
[kokoro-sisyphus_resultstore_prod_foss-fpga-tools_verilog-to-routing_upstream_presubmit_nightly_df5b5b72-664d-4bd7-94e0-b8798e5d65a0_build.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6084208/kokoro-sisyphus_resultstore_prod_foss-fpga-tools_verilog-to-routing_upstream_presubmit_nightly_df5b5b72-664d-4bd7-94e0-b8798e5d65a0_build.log)

sfkhalid, 2021-03-09:
As discussed in the last symbiflow meeting, I added a command in the CI script to gzip the output .net files from the vtr_reg_nightly tests in #1678. After adding this command, the disk space of vtr_flow/tasks/ was reduced from 87G to 65G. 
The full breakdown of space in vtr_flow/tasks/ can be seen in the log file below
[nightly_disk_space_after_gzip.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6109354/nightly_disk_space_after_gzip.txt)

To further reduce space, I can try to also gzip other output files besides .net files. How much of a reduction should we aim for in total @litghost? Would it be best to get the disk space under 50G from the root directory?

litghost, 2021-03-09:
> If there's more low hanging fruit, we should zip more in my opinion.

Great minds et all, see https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1678/files#r590454170

sfkhalid, 2021-03-10:
After zipping all of the files from nightly, the vtr_flow/tasks/ directory is reduced to 3.7G. It took 15 minutes to zip all the files.

litghost, 2021-03-10:
> 
> 
> After zipping all of the files from nightly, the vtr_flow/tasks/ directory is reduced to 3.7G. It took 15 minutes to zip all the files.

I think that is a good trade off, 15 minutes into a 17 hr run is probably fine.

litghost, 2021-03-10:
>  Is the job killed if storage exceeds 100 GB at any point

No.  The 100 GB requirement is once the job is completed.  Basically kokoro copies the working space off of the VM and then kills the VM.  If that copy process exceeds 100 GB, then the job will fail.

vaughnbetz, 2021-03-10:
OK, that sounds OK given what @litghost said.  We'd get more parallelism if we zipped each directory as the regtest completed so we might hide the 15 minute zip time (distribute it across multiple machines, so still worth considering if it's easy, but not essential.

mithro, 2022-02-24:
Is this issue still relevant with the move to GitHub Actions?

mithro, 2022-02-24:
Given that it was from early 2021 and we are now in 2022, I'm going to close it.

closed 2022-02-24 by mithro