verilog-to-routing/vtr-verilog-to-routing #1670: [ODIN] Segmentation fault seen at the top-level of the design, all submodules are fine

aman26kbm, 2021-02-21:
I'm seeing a segmentation fault in the ODIN stage when running VTR on a design.

With a debug build, I see this summary:
SUMMARY: AddressSanitizer: heap-buffer-overflow /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/ast_elaborate.cpp:1387 in finalize_ast(ast_node_t*, ast_node_t*, sc_hierarchy*, bool, bool)

The design is: https://github.com/aman26kbm/vtr-verilog-to-routing/blob/master/vtr_flow/benchmarks/verilog/ml_benchmarks/subunits/compare_fp16_vtr.v

The design is pretty small and has a few components. I tried to do some search to see which part of the design causes the failure. Turns out it's the top-level of the design. Each of the sub-modules pass through the VTR flow without any problem. The top-level has very little logic actually. It just stitches the sub-modules together. I don't know what's going wrong there.

The arch is: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/COFFE_22nm/agilex_arch.xml

Just to add.. Seeing this error at the master of the main VTR repo.

@sdamghan , could you please help identify what the issue is?

aman26kbm, 2021-02-23:
I was trying some more experiments today. 

Here's the hierarchy of the design:
top_module
|--module1 
|  |--submodule1
|  |--submodule2
|-module2 
|  |--submodule3
|  |--submodule4

All modules and submodules use parameters.
I did some binary search and only when the parameters are provided when instantiating module2 in the top-module, the error shows up. At all other places, parameters work fine.
So, I removed params on module1 and hardcoded them. 

So, we're good for now. @sdamghan , you can reduce the priority of this ticket since we have a workaround.
I am attaching the design file that showed the failure (coz the link above will contain the latest modified version when you get to it). 

[compare_fp16_vtr.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6031551/compare_fp16_vtr.zip)

dlrauch15, 2021-02-26:
I have encountered a similar issue. After some debugging I determined that it only happens in this scenario: module A instantiates module B which instantiates module C; B and C have parameters with the _same name_, and the parameter in B is passed to the parameter on C when instantiating it.

Here is some minimal code that reproduces the issue:
```
module mod2 (
   input  wire in,
   output wire out
);
   parameter X = 1'b1;
   assign out = in;
endmodule

module mod1 (
   input  wire in,
   output wire out
);
   parameter X = 1'b1;
   mod2
   #(.X(X))
   i_mod2(
      .in(in),
      .out(out)
   );
endmodule

module top(
   input  wire top_in,
   output wire top_out
);

   mod1 i_mod1(
      .in(top_in),
      .out(top_out)
   );

endmodule
```
Interestingly, the segfault does not occur if module "top" is removed and "mod1" becomes the top module. It also does not occur if I do not "connect" parameter X in mod1 to the mod2 instantiation and instead connect it to a constant or leave it unspecified.

closed 2025-05-25 by github-actions[bot]