verilog-to-routing/vtr-verilog-to-routing #1680: power report is mot generated

rourabpaul1986, 2021-03-04:
I am working voltage scaling technology in FPGA. I was exploring VTR but I am unable to generate the power report. I am using below commands
vtrROOT/vtr_flow/scripts/run_vtr_flow.pl blink.v vtrROOT/vtr_flow/arch/timing/EArch.xml -track_memory_usage -power -cmos_tech vtrROOT/vtr_flow/tech/PTM_22nm/22nm.xml --pack --place.
I got place, route and net report, but not getting power report.
More over I would like to know where to mention the clock frequency
I am using ubuntu 16.06  and vtr-verilog-to-routing-8.0.0

jgoeders, 2021-03-06:
If you specify `--pack --place` then only packing and placement will be run and then it will stop.  Power estimation requires the full flow to be run, so just remove these arguments and then it should work.

closed 2021-03-06 by jgoeders

rourabpaul1986, 2021-03-06:
I understood, I think the blink.v has some issue, because the same commands
are working on other other verilog file. I have solved the problem. Thank
you for the report.
I want to report another issue. The VTR flow does not support input output
declaration in the module of sub circuit.
Let us have two example codes : code 1 and code 2. The  code 1 will not
have any issue if you run it in Odin II, but code 2 will not be running in
odin II, it will  give you core dump error without showing any error
message.
------------------------------------------code
1--------------------------------------------------
module SYNCHRO (input ASYNC, output SYNC, input CLOCK);
  wire C1_ASYNC;
  DFF DFF1(
     .D(ASYNC),
    .CLK(CLOCK),
    .Q(C1_ASYNC)
  );
  DFF DFF2(
    .D(C1_ASYNC),
    .CLK(CLOCK),
   .Q(SYNC)
  );
endmodule

*module DFF ( D, CLK, Q);  input D;  input CLK;*
  *output reg Q;*
  always @ (posedge CLK)
    Q <= D;
endmodule
 -------------------------------------------------------------------------------------
------------------------------------------code
2--------------------------------------------------
module SYNCHRO (input ASYNC, output SYNC, input CLOCK);
  wire C1_ASYNC;
  DFF DFF1(
     .D(ASYNC),
    .CLK(CLOCK),
    .Q(C1_ASYNC)
  );
  DFF DFF2(
    .D(C1_ASYNC),
    .CLK(CLOCK),
   .Q(SYNC)
  );
endmodule

*module DFF ( input D, input CLK, output Q); reg Q;*
  always @ (posedge CLK)
    Q <= D;
endmodule
 -------------------------------------------------------------------------------------

*____________________________________________________________________Dr.
Rourab Paul*
|Assistant Professor | Computer Science and Engineering Dept  |
|SOA Deemed to be University |2nd Floor,   Room No. C227   | Odisha  |
| Phone : +91-7980185683,  Bhubaneswar, Odisa , India |
|Homepage - www.rourab.com mail: rourabpaul@ <rourab.paul@cern.ch>soa.ac.in|
*____________________________________________________________________*

jgoeders, 2021-03-09:
You should probably open another ticket for that. I don't do any work on the Odin tool.