verilog-to-routing/vtr-verilog-to-routing #1691: [QUESTION]: when the VTR project started?

rodrigomelo9, 2021-03-20:
Hi. I want to add basic VTR's info at https://hdl.github.io/awesome/items/vtr/ and I am not sure about when the project started. The first commit in this repository is from 2011, but I found copyright notices from years before. Also useful to know when VPR started, to be documented at https://hdl.github.io/awesome/items/vpr.

Thanks,
Rodrigo

rodrigomelo9, 2021-03-20:
I opened a draft PR (https://github.com/hdl/awesome/pull/184) to update VTR and VPR info at [hdl/awesome](https://github.com/hdl/awesome). Would be great a review from people involved in the project, to ensure the quality of the provided data.

mithro, 2021-03-20:
FYI -- It is my understanding that VPR was part of Vaughn Betz's PhD. He then started the company RightTrackCAD which was acquired by Altera to become Quartus II. Vaughn let Altera at some point, came back to the University of Toronto, dusted off the original VPR and created VtR.

mithro, 2021-03-20:
V. Betz and J. Rose, "VPR: A New Packing, Placement and Routing Tool for FPGA Research,'' Seventh International Workshop on Field-Programmable Logic and Applications, London, UK, 1997, pp. 213 - 222.

mithro, 2021-03-20:
https://www.eecg.utoronto.ca/~vaughn/publications.html

rodrigomelo9, 2021-03-22:
Hi @mithro, so VPR is around 1996/7. Thanks for the info and the provided links. I am still not sure about dates, but for sure before 2011 :-D I will try to found when VtR started. Regards

tangxifan, 2021-03-24:
@rodrigomelo9 You may refer to the https://janders.eecg.utoronto.ca/pdfs/p77-rose.pdf which is the 1st paper that announces the VTR project. It was published in FPGA'2012. 

Before that, only VPR codes are released. As @mithro said, the first version is VPR 4.3 which is published at FPL'1997.
Some major update are
- VPR5 which is published in FPGA'2009 
- VPR6 which is published in FPGA'2011
- VPR7 which is published in FPGA'2012

rodrigomelo9, 2021-03-25:
Great! The paper that you mentioned is from February 2012, so VTR is probably from 2011, which is where the repository was initialized :-D

Thanks! I think that it was solved (I will use 1997 for VPR and 2011 for VtR).

Regards,
Rodrigo

closed 2025-05-16 by AmirhosseinPoolad