verilog-to-routing/vtr-verilog-to-routing #1699: [ODIN] Remove the requirement that module instantiation port assignment order must match port order in module definition.

dlrauch15, 2021-03-29:
Right now, ODIN requires that port assignments in module instantiations must match the order in which the ports are defined in their module definitions. While this is a simple requirement to satisfy for hand-written RTL designs, it is a challenge when dealing with large HLS-generated designs that can be 10s of thousands of lines long with 10s to 100s of module instantiations that do not obey this rule.

@sdamghan @aman26kbm

sdamghan, 2021-03-30:
@dlrauch15 , would you please attach a small Verilog sample code plus relevant input/output files? If memory helps me this wasn't a problem. Odin can only do either port connection by name OR by ordered list. However, it cannot perform the combination of them.
```
module module_x (input a, output q);
---------------------------------------------------------
module_x uut (.in(a), .out(q) ) -> working
module_x uut (.out(q), .in(a) ) -> working
module_x uut ( a, q ) -> working
module_x uut (.in(a), q ) -> does not work
```

aman26kbm, 2021-04-11:
Hi @sdamghan , just wanted to clarify. The issue is not with port connection by name or ordered list. The issue is with the requirement of matching the order in which the ports are defined in the module definition and the order in which they are provided on the module instantiation.

Example:
Let's say this is the module definiton:
module mymodule(
input x,
input y,
output z);
...
endmodule

Or it is this:
module mymodule(x,y,z);
input x;
input y;
output z;
...
endmodule

While instantiating:

If we use:
mymodule inst (.x(), .y(), .z())
then it works.

But if we use:
mymodule inst(.z(), .y(), .x())  //Note that the order is not x,y,z
then it fails.

@dlrauch15 please correct me if I am not explaining it correctly.

dlrauch15, 2021-04-11:
That is correct but it sounds like Seyed is saying he believes that works (see the second line in his provided code). I will investigate this more next week.

dlrauch15, 2021-04-12:
Hi @sdamghan,

I confirmed that mixed port instantiation order works for simple cases, but not more complex ones. Here is a minimal test case that reproduces the issue:

```
module m (
   output wire o1,
   input wire i1,
   input wire i2,
   output wire o2
);
   assign o1 = ~i1;
   assign o2 = ~i2;
endmodule

module top (
   input wire i1,
   input wire i2,
   output wire o1,
   output wire o2
);

   m m_inst (
     .i1(i1),
     .i2(i2),
     .o2(o2),
     .o1(o1)
   );

endmodule
```
The problem goes away if I remove any of the four ports, or if I reorder the ports in some ways (I did not try every possible order, but this one fails).

The error message I get is:
```
/home/daniel/research/main_repo/vtr-verilog-to-routing/vtr_work/scratch/test.v:2:2: error[NETLIST]:  This module entry does not match up correctly (i2 != o1).  Odin expects the order of ports to be the same
     1 | 
     2 |    output wire o1,
           ^~~~
     3 |    input wire i1,
     4 |    input wire i2,

/home/daniel/research/main_repo/vtr-verilog-to-routing/ODIN_II/SRC/netlist_create_from_ast.cpp:1969: void connect_module_instantiation_and_alias(short int, ast_node_t*, char*, sc_hierarchy*): Fatal error
./testrun.sh: line 1:  9771 Aborted                 (core dumped) $VTR_ROOT_MAIN/ODIN_II/odin_II -a $VTR_ROOT_MAIN/vtr_flow/arch/COFFE_22nm/agilex_arch.xml -V test.v -o test.blif
```
Let me know if there's any other information I can provide. Thanks!

sdamghan, 2021-04-13:
Thanks, @dlrauch15 , This is quite odd behaviour. If the output is not defined at the first line, the error will disappear. This happens while I have tested other files like this and did not see anything similar. Anyway, this is a bug; let me go through it, and I will let you know.

closed 2021-05-08 by sdamghan