verilog-to-routing/vtr-verilog-to-routing #1703: Odin: Default data type should be wire

harpreetbamrah, 2021-04-06:
#### Expected Behaviour
Data type should be inferred as wire if not specified

#### Current Behaviour
Odin does not default data type to wire when there is no wire/reg definiton

#### Possible Solution

#### Steps to Reproduce
```
module simple_op(a,b,c);

input a;
input b;

output c;

always @(*) begin
    c = a & b;
end

endmodule
```

#### Context
Happened while investigating #1410 

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

harpreetbamrah, 2021-04-16:
may refer to object "out" in regression_test/benchmark/verilog/syntax/multi_clock_reader_writer.v

vaughnbetz, 2021-04-22:
@seyed: I suspect you're the bet person to look into this.

sdamghan, 2021-04-22:
Thanks @vaughnbetz ,
We are actually working on this issue on the CASA fork. Will make a PR and let you know when it's done.