verilog-to-routing/vtr-verilog-to-routing #1721: Inconsistency between post synthesis BLIF and Verilog netlists

mkurc-ant, 2021-05-07:
VPR can write post synthesis netlist (it is more of a post-pnr netlist IMO) as BLIF and Verilog. In the BLIF file unconnected input and output ports are tied to special nets named `__vpr__unconnXX` which is fine. However, in the Verilog file they are not represented correctly.

#### Expected Behaviour
In post synthesis netlist in Verilog format:
 - unconnected input cell ports should be unconnected or tied to `1'bX` or tied to a special net named `__vpr__unconnXX`.
 - unconnected output cell ports should be either unconnected or tied to a special net named `__vpr__unconnXX`.

#### Current Behaviour
In post synthesis netlist in Verilog format:
 - unconnected input cell ports are always tied to `1'b0` which is clearly incorrect
 - all unconnected output cell ports are tied to the same net named `DummyOut` which is even more incorrect as it results in multi-source nets.

#### Possible Solution
Fix this part of the code https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/a71ca6d2d755ccc372bcbe12faa65d3a3ac5788b/vpr/src/base/netlist_writer.cpp#L2188-L2196 to achieve the desired behavior

#### Steps to Reproduce
For any architecture and design run the full VPR flow (pack, place and route) with `--analysis` and `--gen_post_synthesis_netlist on` options.

#### Context

I want the output post-synthesis netlist in Verilog format to be usable for design simulation.

#### Your Environment

* VTR revision used: `8.0.0-3452-ge7d45e013` (same issue would occur with the current master)
* Operating System and version: Linux Ubuntu 18.04
* Compiler version: `GNU 7.3.0 on Linux-4.15.0-1109-azure x86_64`

vaughnbetz, 2021-05-13:
Discussed in meeting; agreed that we can copy the blif behaviour so long as it doesn't confuse downstream verilog simulators.
- Also test this to make sure this works with a simple design and a post-routing simulation.
- Suggestion: allow these unconnect inputs to be driven to 0, 1 or X (leave unconnected) by default.

vaughnbetz, 2025-05-25:
This was fixed long ago and should have been closed; leaving it closed.

closed 2025-05-25 by vaughnbetz