verilog-to-routing/vtr-verilog-to-routing #1727: Broken documentation links in read_the_docs, but not in direct viewing of .md files on github

vaughnbetz, 2021-05-13:
Most VTR documentation is generated with read the docs, but a few key pieces of documentation are in .md files in the root directory of the vtr repo (e.g. README.developers.md, CONTRIBUTING.md, SUPPORT.md).  You can see a full list at https://github.com/verilog-to-routing/vtr-verilog-to-routing

The read the docs documentation has links to some of these .md files, and they aren't working (get a file not found error).  The links do work if you are viewing the CONTRIBUTING.md file or README.developers.md file directory on github (browsing the repo), but not if you click on them in the generated read the docs documentation.  

For example, from the Developer Guide/Contribution Guidelines page on read the docs (https://docs.verilogtorouting.org/en/latest/dev/contributing/contributing/) click on the "Support Resources" link.  It is broken.  But this same link works if you directly read CONTRIBUTING.md in the github repo.

The same issue exists in Developer Guide/Commit Procedures page in read the docs for the "Submtting Code to VTR" link.  It is broken in read the docs, but works if you directly read README.developers.md in the github repo.

#### Expected Behaviour
The links should work in read the docs (and should still work if you directly browse the github repo).

#### Current Behaviour
Links work if you directly browse the relevant .md file, but not in read the docs.  Note that most links in read the docs work; it's just these special ones that link to .md files in the vtr root directory that don't.

#### Possible Solution
Not sure; hopefully there is some syntax that lets these links work in both cases.  If not, perhaps we should move to only supporting read the docs (i.e. put these files in the usual spot?).

#### Steps to Reproduce
Go to the vtr documentation (e.g. https://docs.verilogtorouting.org/en/latest/dev/developing/), and click on the links listed above (e.g. "Submitting Code to VTR").

#### Context
It's annoying; I thought the links were broken while using the normal browser interface to the VTR documentation.

#### Your Environment
Reading through chrome on Windows.

mithro, 2021-05-13:
FYI -- @acomodi and @rw1nkler

vaughnbetz, 2021-06-18:
Any ideas @acomodoi and @rw1nkler ?

mithro, 2021-06-18:
@kgugala - Can you poke @acomodi and @rw1nkler to look into this ASAP.

acomodi, 2021-06-18:
Sorry, lost track of this. I'll take a look. Possibly there is an issue with the recommonmark library.

closed 2021-06-24 by acomodi