verilog-to-routing/vtr-verilog-to-routing #1738: VNumber copy constructor with a new length does set defined_size attribute

sdamghan, 2021-05-25:
#### Expected Behaviour

When a new VNumber is created using a copy constructor with a new length, the `defined_size` value should set to `true`. 

#### Current Behaviour

A new VNumber is created with a new size while the `defined_size` is false. 

#### Possible Solution

Creating a new VNumber using copy constructor with a new length, the `defined_size` should set to `true`.

#### Steps to Reproduce

1. Run Odin_II with the following test case:
`${VTR_ROOT}/ODIN_II/odin_ii -V` [minus_operator.v](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6535961/minus_operator.v.txt)

#### Context

#### Your Environment

* VTR revision used: 8.1.0-dev+00617ff76
* Operating System and version: 20.04.1-Ubuntu SMP
* Compiler version: GNU 9.3.0 on Linux-5.8.0-53-generic x86_64

ArashAhmadian, 2021-05-28:
@sdamghan PR #1739 didn't fix the concatenation issue we discussed in email. 
[image]
It has the same behaviour as prior to the PR.

sdamghan, 2021-05-28:
@ArashAhmadian That's weird! I've just checked the pipelined_fft_64.v_ with Odin_II, and it worked properly.
Would you compile again and run the following command in the ODIN_II directory? Please let me know the result.
`./odin_II -V ../vtr_flow/benchmarks/arithmetic/generated_circuits/opencores/verilog/pipelined_fft_64.v`

ArashAhmadian, 2021-05-28:
> @ArashAhmadian That's weird! I've just checked the pipelined_fft_64.v_ with Odin_II, and it worked properly.
> Would you compile again and run the following command in the ODIN_II directory? Please [...]

@sdamghan I had forgotten to compile ODIN after pulling :) (sorry) . It synthesizes properly based on the results of the command you sent.  However, when running the open_core task an assertion fails: 
```
/home/ahmadi55/vtr-verilog-to-routing/ODIN_II/SRC/subtractions.cpp:467: split_adder_for_sub: Assertion not_node[index]->input_pins[0]->net->num_driver_pins == 1 failed
   465 |                      * if the input pin of not gate connects to vcc, replacing the input pin and the not gate with gnd.*/
   466 |                     const int index = i * sizeb + j - offset;
   467 |                     oassert(not_node[index]->input_pins[0]->net->num_driver_pins == 1);
        ^~~~
   468 |                     if (not_node[index]->input_pins[0]->net->driver_pins[0]->node->type == GND_NODE) {
   469 |                         connect_nodes(netlist->vcc_node, 0, node[i], (sizea + j));
Command terminated by signal 6
```
It is weird how it synthesizes properly when running Odin on its own but fails when running it through flow.

sdamghan, 2021-05-28:
@ArashAhmadian no worries, the above issue is different from the previous one. This one caused by the architecture file, not the Verilog file. I will investigate more on this, would you please create another issue on this matter? Thanks

ArashAhmadian, 2021-05-28:
@sdamghan Will do. Thanks

closed 2021-07-05 by sdamghan