verilog-to-routing/vtr-verilog-to-routing #1745: Fails to build with GCC 11.1

Xiretza, 2021-05-28:
#### Expected Behaviour
vtr compiles.

#### Current Behaviour
Compiling vtr with GCC 11.1 fails:

```
In file included from /usr/include/c++/11.1.0/bits/stl_algobase.h:59,
                 from /usr/include/c++/11.1.0/vector:60,
                 from /build/vtr-git/src/vtr-verilog-to-routing/libs/libvtrutil/src/vtr_linear_map.h:3,
                 from /build/vtr-git/src/vtr-verilog-to-routing/vpr/src/timing/PostClusterDelayCalculator.h:3,
                 from /build/vtr-git/src/vtr-verilog-to-routing/vpr/src/timing/AnalysisDelayCalculator.h:4,
                 from /build/vtr-git/src/vtr-verilog-to-routing/vpr/src/analysis/timing_reports.h:5,
                 from /build/vtr-git/src/vtr-verilog-to-routing/vpr/src/analysis/timing_reports.cpp:1:
/usr/include/c++/11.1.0/bits/stl_vector.h: In instantiation of ‘void std::vector<_Tp, _Alloc>::swap(std::vector<_Tp, _Alloc>&) [with _Tp = t_rr_node_data; _Alloc = vtr::aligned_allocator<t_rr_node_data>]’:
/usr/include/c++/11.1.0/bits/allocator.h:308:28:   required from ‘static bool std::__shrink_to_fit_aux<_Tp, true>::_S_do_it(_Tp&) [with _Tp = std::vector<t_rr_node_data, vtr::aligned_allocator<t_rr_node_data> >]’
/usr/include/c++/11.1.0/bits/vector.tcc:693:56:   required from ‘bool std::vector<_Tp, _Alloc>::_M_shrink_to_fit() [with _Tp = t_rr_node_data; _Alloc = vtr::aligned_allocator<t_rr_node_data>]’
/usr/include/c++/11.1.0/bits/stl_vector.h:990:9:   required from ‘void std::vector<_Tp, _Alloc>::shrink_to_fit() [with _Tp = t_rr_node_data; _Alloc = vtr::aligned_allocator<t_rr_node_data>]’
/build/vtr-git/src/vtr-verilog-to-routing/vpr/src/route/rr_graph_storage.h:442:36:   required from here
/usr/include/c++/11.1.0/bits/stl_vector.h:1483:9: error: no match for ‘operator==’ (operand types are ‘std::_Vector_base<t_rr_node_data, vtr::aligned_allocator<t_rr_node_data> >::_Tp_alloc_type’ {aka ‘std::allocator_traits<vtr::aligned_allocator<t_rr_node_data> >::rebind_alloc<t_rr_node_data>’} and ‘std::_Vector_base<t_rr_node_data, vtr::aligned_allocator<t_rr_node_data> >::_Tp_alloc_type’ {aka ‘std::allocator_traits<vtr::aligned_allocator<t_rr_node_data> >::rebind_alloc<t_rr_node_data>’})
 1483 |         __glibcxx_assert(_Alloc_traits::propagate_on_container_swap::value
      |         ^~~~~~~~~~~~~~~~
```

#### Possible Solution

This is caused by: https://github.com/gcc-mirror/gcc/commit/10f51543bb81cc953792270b40a9c812049e8b4c

`__glibcxx_assert` now includes a compile-time static assertion even if `_GLIBCXX_ASSERTIONS` are disabled, so the [assert in `std::vector::swap`](https://github.com/gcc-mirror/gcc/blob/releases/gcc-11.1.0/libstdc%2B%2B-v3/include/bits/stl_vector.h#L1483-L1484) now fails to compile for [`t_rr_graph_storage::node_storage_`](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/2a2ae0f1dbbdd4a2dc9e85a98c518d6820437002/vpr/src/route/rr_graph_storage.h#L650) (because `vtr::aligned_allocator` has no `operator==`). A possible solution might be to implement `vtr::aligned_allocator::operator==`.

#### Steps to Reproduce
1. Install GCC 11.1
2. Build vtr 2a2ae0f1dbbdd4a2dc9e85a98c518d6820437002

#### Your Environment
* VTR revision used: 2a2ae0f1dbbdd4a2dc9e85a98c518d6820437002
* Operating System and version: Arch Linux
* Compiler version: GCC 11.1

vaughnbetz, 2021-05-28:
Thanks for flagging this.  Since you have a proposed solution I suspect you're in the best position to attempt a fix, if you have the time to look into this further.

Xiretza, 2021-05-28:
I can try to look into it, but I don't really know anything about VTR internals, so it will probably take me a while to get something working.

vaughnbetz, 2021-05-28:
Thanks.  Adding @MohamedElgammal as he knows the vtr_vector code so perhaps he has a thought too.

PR #1749 (2021-06-01): aligned_allocator: add operator==
#### Description
Fixes #1745, see the commit message or the issue for details.

#### How Has This Been Tested?
The test suite still passes, and compilation is now successful with GCC 11.

#### Types of changes
- [x] Bug fix (change which fixes an issue)
- [ ] New feature (change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

#### Checklist:
- [ ] My change requires a change to the documentation
- [ ] I have updated the documentation accordingly
- [ ] I have added tests to cover my changes
- [ [... cut]
closed 2021-06-01 by vaughnbetz