verilog-to-routing/vtr-verilog-to-routing #1765: Move to newer clang-format

mithro, 2021-06-04:
#### Expected Behaviour

Currently we use `clang-format-7` on Verilog to Routing. [`clang-format` is now up to version 13.](https://clang.llvm.org/docs/ClangFormat.html).

clang also helpfully provides precompiled versions of its tools for all major currently supported apt based distros at https://apt.llvm.org/

Given that `clang-format-11` is considered the current "oldstable" version. I suggest it is safe for us to move to using that.

mithro, 2021-06-04:
@Xiretza want to take a look?

Xiretza, 2021-06-11:
Currently, Ubuntu 20.04 is the [latest available image](https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions#github-hosted-runners) for GHA. Unless we add new apt repositories (I'm not sure what the cleanest way to do that would be), the best we can get is clang 10: https://packages.ubuntu.com/search?suite=focal&arch=any&searchon=names&keywords=clang-format

mithro, 2021-06-18:
From https://apt.llvm.org/
`bash -c "$(wget -O - https://apt.llvm.org/llvm.sh)"`

See also `wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key|sudo apt-key add -` and
```
# i386 not available
deb http://apt.llvm.org/focal/ llvm-toolchain-focal main
deb-src http://apt.llvm.org/focal/ llvm-toolchain-focal main
# 11
deb http://apt.llvm.org/focal/ llvm-toolchain-focal-11 main
deb-src http://apt.llvm.org/focal/ llvm-toolchain-focal-11 main
# 12
deb http://apt.llvm.org/focal/ llvm-toolchain-focal-12 main
deb-src http://apt.llvm.org/focal/ llvm-toolchain-focal-12 main
```

AmirhosseinPoolad, 2025-05-16:
Closing since we're using the latest clang-format that is available on Ubuntu 24.04 now.

closed 2025-05-16 by AmirhosseinPoolad