verilog-to-routing/vtr-verilog-to-routing #1767: Add doxygen documentation of atom_netlist and clb_netlist to vpr API web documentation

vaughnbetz, 2021-06-05:
The base netlist class is documented in the vpr api documentation in read the docs, but the derived atom and clb netlists are not. Those netlists have doxygen-compatible comments already, so they should be added to the doxygen build settings in the documentation so they show up in the read-the-docs documentatin.

#### Proposed Behaviour
Doxygen comments (which already exist) should create new vpr api documentation on the vtr web page as sub-headings under the netlist documentation.

#### Current Behaviour
Currently the base netlist class has auto-generated documentation, but these derived classes do not.

#### Possible Solution
Update the Doxygen configuration to add these two classes.

#### Context
To ramp up new developers and expand our documentation we should get this on the web docs.

mithro, 2021-06-06:
@rw1nkler - Any idea why these are not showing up?

rw1nkler, 2021-06-10:
> @rw1nkler - Any idea why these are not showing up?

The derived classes need to be specified in the proper sphinx RST file, in the similar way as it is done for the base `Netlist` class:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/doc/src/api/vpr/netlist.rst#classes

As it was stated, those files should be already parsed by Doxygen and the references to them should be available in Sphinx.

vaughnbetz, 2021-06-10:
@rw1nkler or @MohamedElgammal : can one of you add the two derived classes in the appropriate spot?

vaughnbetz, 2021-06-18:
@rw1nkler @MohamedElgammal : nag, nag, nag ...

It would be really nice to have these in the readthedocs pages, as I'd like to point some new developers at them.

MohamedElgammal, 2021-06-18:
@vaughnbetz will handle this and mention you in the corresponding PR.

vaughnbetz, 2021-06-18:
Thanks @MohamedElgammal

closed 2021-06-21 by MohamedElgammal