verilog-to-routing/vtr-verilog-to-routing #1782: Add Koios benchmark QoR example to the vtr developer guide, and add koios to CI

vaughnbetz, 2021-06-17:
The Koios benchmarks have been integrated into the vtr repo, but to have the most impact, we should do two more things:

1. Add a short description (and links to the benchmark README) to the VTR developer guide under the QoR Collection section: https://docs.verilogtorouting.org/en/latest/dev/developing/#collecting-qor-measurements
It should come after the Titan benchmarks (or maybe after the VTR benchmarks if that flows better as they go all the way through odinII ...)_ and have a similar style.

2. The vtr_reg_nightly4 suite that runs them should be added to CI.  @ArashAhmadian can work with you to make the necessary changes and pin Tim Ansell to make the eventual google-side changes.

aman26kbm, 2021-06-17:
Thanks for filing this, Vaughn. 

I will add to task #2. I need to add protection of benchmarks without the "\`define complex_dsp". And I need to update the documentation to mention how to run benchmarks with and without complex_dsp (the new include file method). 

I will work on these within next week.

aman26kbm, 2021-06-23:
I've created a new PR which includes the changes for task #1 mentioned in this issue.

@ArashAhmadian , can you please guide me on what changes to make changes for task #1?

ArashAhmadian, 2021-06-23:
@aman26kbm Sure,  do you mean changes to add vtr_reg_nightly_test4 to CI? Your previous PR actually added the .cfg file and made the necessary changes in vtr_test.sh so I think  everything is good from our side.

aman26kbm, 2021-06-23:
Right. That's great.
So, that means the Koios tests have been running in nightly regressions over the last week?

ArashAhmadian, 2021-06-23:
I don't see them being run. I think @mithro needs to add the configs on the Google side of things for them to start running.

aman26kbm, 2021-06-23:
Ok. Thanks. We can discuss this with @mithro in this week's VTR meeting on Thursday.

ArashAhmadian, 2021-06-23:
Sounds good.

ArashAhmadian, 2021-06-24:
@aman26kbm What's your email? I am drafting the email to Arnaud and I can put you on it.

vaughnbetz, 2021-06-24:
Aman is at aman.kbm@utexas.edu

vaughnbetz, 2025-05-16:
This has been completed; marking closed.

closed 2025-05-16 by vaughnbetz