verilog-to-routing/vtr-verilog-to-routing #1788: Build fails due to errors in `Hashable.cpp`

siddhpant, 2021-06-24:
Ran `make CMAKE_PARAMS="-DVTR_ENABLE_SANITIZE=ON"`

#### Expected Behaviour
`make` should have been successful.

#### Current Behaviour
It fails, spitting the following:

```
Scanning dependencies of target libodin_ii
[ 98%] Building CXX object ODIN_II/CMakeFiles/libodin_ii.dir/SRC/Hashtable.cpp.o
In file included from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:27:
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:33:26: error: ‘string’ is not a member of ‘std’
   33 |  std::unordered_map<std::string,void*> my_map;
      |                          ^~~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:29:1: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
   28 | #include <unordered_map>
  +++ |+#include <string>
   29 | 
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:33:26: error: ‘string’ is not a member of ‘std’
   33 |  std::unordered_map<std::string,void*> my_map;
      |                          ^~~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:33:26: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:33:38: error: template argument 1 is invalid
   33 |  std::unordered_map<std::string,void*> my_map;
      |                                      ^
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:33:38: error: template argument 3 is invalid
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:33:38: error: template argument 4 is invalid
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:33:38: error: template argument 5 is invalid
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:38:34: error: ‘std::string’ has not been declared
   38 |  void   add                (std::string key, void *item);
      |                                  ^~~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:40:9: error: expected ‘;’ at end of member declaration
   40 |  void*  remove             (std::string key);
      |         ^~~~~~
      |               ;
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:40:40: error: expected ‘)’ before ‘key’
   40 |  void*  remove             (std::string key);
      |                            ~           ^~~~
      |                                        )
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:42:9: error: expected ‘;’ at end of member declaration
   42 |  void*  get                (std::string key);
      |         ^~~
      |            ;
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:42:40: error: expected ‘)’ before ‘key’
   42 |  void*  get                (std::string key);
      |                            ~           ^~~~
      |                                        )
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp: In member function ‘void Hashtable::destroy_free_items()’:
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:33:15: error: ‘begin’ was not declared in this scope; did you mean ‘std::begin’?
   33 |  for(auto kv: my_map)
      |               ^~~~~~
      |               std::begin
In file included from /usr/include/c++/10/array:41,
                 from /usr/include/c++/10/tuple:39,
                 from /usr/include/c++/10/bits/hashtable_policy.h:34,
                 from /usr/include/c++/10/bits/hashtable.h:35,
                 from /usr/include/c++/10/unordered_map:46,
                 from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:28,
                 from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:27:
/usr/include/c++/10/bits/range_access.h:108:37: note: ‘std::begin’ declared here
  108 |   template<typename _Tp> const _Tp* begin(const valarray<_Tp>&);
      |                                     ^~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:33:15: error: ‘end’ was not declared in this scope; did you mean ‘std::end’?
   33 |  for(auto kv: my_map)
      |               ^~~~~~
      |               std::end
In file included from /usr/include/c++/10/array:41,
                 from /usr/include/c++/10/tuple:39,
                 from /usr/include/c++/10/bits/hashtable_policy.h:34,
                 from /usr/include/c++/10/bits/hashtable.h:35,
                 from /usr/include/c++/10/unordered_map:46,
                 from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:28,
                 from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:27:
/usr/include/c++/10/bits/range_access.h:110:37: note: ‘std::end’ declared here
  110 |   template<typename _Tp> const _Tp* end(const valarray<_Tp>&);
      |                                     ^~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp: At global scope:
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:37:6: error: no declaration matches ‘void Hashtable::add(std::string, void*)’
   37 | void Hashtable::add(std::string key, void *item)
      |      ^~~~~~~~~
In file included from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:27:
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:38:9: note: candidate is: ‘void Hashtable::add(int, void*)’
   38 |  void   add                (std::string key, void *item);
      |         ^~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:30:7: note: ‘class Hashtable’ defined here
   30 | class Hashtable
      |       ^~~~~~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:51:7: error: no declaration matches ‘void* Hashtable::remove(std::string)’
   51 | void* Hashtable::remove(std::string key)
      |       ^~~~~~~~~
In file included from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:27:
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:40:9: note: candidate is: ‘void* Hashtable::remove’
   40 |  void*  remove             (std::string key);
      |         ^~~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:30:7: note: ‘class Hashtable’ defined here
   30 | class Hashtable
      |       ^~~~~~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:63:7: error: no declaration matches ‘void* Hashtable::get(std::string)’
   63 | void* Hashtable::get(std::string key)
      |       ^~~~~~~~~
In file included from /home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:27:
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:42:9: note: candidate is: ‘void* Hashtable::get’
   42 |  void*  get                (std::string key);
      |         ^~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/include/Hashtable.hpp:30:7: note: ‘class Hashtable’ defined here
   30 | class Hashtable
      |       ^~~~~~~~~
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp: In member function ‘bool Hashtable::is_empty()’:
/home/yoojer/.local/VTR/vtr-verilog-to-routing-8.0.0/ODIN_II/SRC/Hashtable.cpp:75:17: error: request for member ‘size’ in ‘((Hashtable*)this)->Hashtable::my_map’, which is of non-class type ‘int’
   75 |  return (my_map.size() == 0);
      |                 ^~~~
make[3]: *** [ODIN_II/CMakeFiles/libodin_ii.dir/build.make:93: ODIN_II/CMakeFiles/libodin_ii.dir/SRC/Hashtable.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:1471: ODIN_II/CMakeFiles/libodin_ii.dir/all] Error 2
make[1]: *** [Makefile:160: all] Error 2
make: *** [Makefile:76: all] Error 2
```
In addition to this, I had encountered numerous warnings earlier in the process. I have lost the full dump since my terminal's buffer/history didn't store everything.

#### Steps to Reproduce

1. Download the latest tar.gz from releases page and extract
2. Run `make CMAKE_PARAMS="-DVTR_ENABLE_SANITIZE=ON"`

#### Context
I was trying to build VTR as listed in the docs.

#### Your Environment

* VTR revision used: 8.0.0
* Operating System and version: SMP Debian 5.10.40-1kali1 (2021-05-31) x86_64 GNU/Linux
* Compiler version:
    * gcc (Debian 10.2.1-6) 10.2.1 20210110
    * Debian clang version 11.0.1-2
    * GNU Make 4.3
    * cmake version 3.18.4

eminfedar, 2021-07-02:
Same. I have tried to build just with `make` and these errors appeared.

My Environment

    VTR revision used: 8.0.0
    Operating System and version: Linux mi 5.10.0-7-amd64 #1 SMP Debian 5.10.40-1 (2021-05-28) x86_64 GNU/Linux
    Compiler version:
        gcc (Debian 10.2.1-6) 10.2.1 20210110
        Debian clang version 11.0.1-2
        GNU Make 4.3
        cmake version 3.18.4

Daedgomez, 2021-08-02:
Did you solve this error?

siddhpant, 2021-08-03:
> Did you solve this error?

No... There were many warnings preceding this, and building upto this point took a loooong time on my machine, so I didn't compile again.

Although it seems like they had addressed this specific issue [back in 2020 itself](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commits/master/ODIN_II/SRC/include/Hashtable.hpp), but the release is older than that so we have the earlier file.

Maybe try compiling the master branch if you are up to.

closed 2025-05-28 by AlexandreSinger