verilog-to-routing/vtr-verilog-to-routing #1790: The Trunk Status QoR Tracking links are not working

kkumar23, 2021-06-25:
#### Expected Behaviour

The links ok the Trunk Status and QoR Tracking redirects to http://builds.verilogtorouting.org:8080/waterfall

#### Current Behaviour

But these seems to be broken or no access given!

#### Possible Solution

Fix the link or provide access

#### Steps to Reproduce

1. Go to homepage  https://github.com/verilog-to-routing/vtr-verilog-to-routing
2.  Under section Development  of readmd, you can find the links to Trunk Status and QoR Tracking.

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

vaughnbetz, 2021-07-05:
We've been moving away from the buildbot tracking to the CI integrated with github, but should either make those links work again or replace with github equivalents.  @MohamedElgammal let's discuss.

vaughnbetz, 2025-05-19:
We don't maintain this build tracking anymore, and the links have been removed from the vtr website.

closed 2025-05-19 by vaughnbetz