verilog-to-routing/vtr-verilog-to-routing #1803: Coordination with hdl/containers

umarcor, 2021-07-18:
I'm about to add Verilog-to-Routing to the container collection in hdl/containers (gcr.io/hdl-containers, ghcr.io/hdl, docker.io/hdlc). As commented in hdl/containers#38, there are some issues which I'm unsure about:

> * [ ]  Should the image name be `vtr` or `vpr`?
> * [ ]  It seems that `PREFIX` is not honored in the build scripts.
> * [ ]  Are there any runtime dependencies apart from libgtk-3-bin?

I'd be glad if maintainers/developers could help me solve those.

vaughnbetz, 2021-08-12:
Commented in https://github.com/hdl/containers/pull/38

umarcor, 2021-08-31:
- The naming was clarified.
- The issue with PREFIX was fixed by using cmake only, instead of make.
- The question about runtime dependencies remains. However, container images are available already (gcr.io/hdl-containers/vtr), so we can handle possibly missing dependencies as bugfixes.

vaughnbetz, 2025-05-19:
This was completed ... closing.

closed 2025-05-19 by vaughnbetz