verilog-to-routing/vtr-verilog-to-routing #1806: mcnc benchmarks in wrong place / missing in regtests

vaughnbetz, 2021-07-22:
Right now there appears to be an mcnc test over a bunch of benchmarks that tests implementation and formal equivalence in the regression_mcnc in vtr_reg_basic, but it isn't in the task list so it isn't running.
This test is probably too big and slow to be in basic anyway, but it shouldn't be in the directory then.

vtr_reg_strong has a smaller mcnc test (only a few circuits).  It doesn't check equivalence.  

The orphaned test in vtr_reg_basic should probably be moved to one of the vtr_nightly tests, and added to the task list.

vaughnbetz, 2021-07-22:
@ArashAhmadian : let me know if this is something you can handle.

ArashAhmadian, 2021-07-22:
@vaughnbetz Sure, I will look into this.

ArashAhmadian, 2021-07-30:
Closing issue.

closed 2021-07-30 by ArashAhmadian