verilog-to-routing/vtr-verilog-to-routing #1808: Runs with sanitization on seems to be running out of memory in github actions CI

vaughnbetz, 2021-07-29:
Approximately 10 days ago runs that use sanitization in CI (S: vtr_reg_basic, odin_basic_tests) started failing with an error code that indicates out of memory in the CI tests.  We also get some QoR test failures on those same designs that indicate peak memory footprint has increased by 14x. This is affecting multiple pull requests, so we haven't been able to track it to any recent code change.

@tangxifan has run the sanitizers locally, and they complete fine, with no unusually high memory footprint.  @sfkhalid : I believe you have seen the same thing (there isn't a problem when you run these locally).

Initially I suspected that the recent rr-graph refactoring could have triggered a bad memory behaviour (large footprint) with the sanitizers, but that doesn't seem to be the case so far. @tangxifan has tried rolling back various changes locally and so far hasn't seen a correlation between his changes and these sanitizer failures.  Also, the Odin II basic tests are failing, and they do not use the router, but do use sanitization.  

The S: vtr_reg_strong tests are larger designs that also use sanitization, and they are working fine. They run on kokoro though, not on github actions.  

I have gone through the github action configuration and cmake files and so far haven't seen any recent changes that seem related at all (or any recent changes, in most cases).  

So the leading theory at this point is that something changed in the github runners that is causing the problem.  github action runners are supposed to have 7 GB of memory available for two cores.  @tangxifan found that the S:vtr_reg_basic tests only used ~400 MB when run locally (one at a time), so this should be fine, even if we double the 400 MB for two parallel runs.

I am a bit baffled, but this is impacting our ability to merge code, so it's important to figure out.  One hacky fix could be to move all sanitized runs (S:vtr_reg_basic, odin_basic_regression) to kokoro, which appears to have no problems.  That isn't too satisfying a solution though.

@kgugala : Tim suggested you were the best person to debug github actions, so assigning this to you. @tangxifan : please summarize your data to date here, and @sfkhalid and @sdamghan please add any additional information you have. @umarcor : did the removal of travis in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1804 lead to any configuration settings that have already been changed that might have caused this?

https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1802 has some earlier discussion on this, and shows the symptom.

vaughnbetz, 2021-07-29:
@sdamghan : this PR https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1807 has a strange behaviour.  It appears Odin reg basic succeeded (which would be odd given the sanitization issues in CI) but when I look at the logs I see multiple sigabrts in Odin II runs, and a 14x vpr_peak_memory increase, which leads to QoR failures.  But the final regression test indicates success anyway !  Can you take a look and see if the regtest is properly checking for all errors before returning success or failure?  This seems like a separate issue from the sanitizer issue, but one that should be fixed too, assuming I'm not misinterpreting something).

tangxifan, 2021-07-29:
Thanks @vaughnbetz for summarizing the issue.

Here are the details:
- I see the sanity checks passed in past when we merge the PRs (even on the current master) but it starts to turn red last Monday when rerun these tests. So, it seems to me that Github runner has changed memory limits.
- I created a branch https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/test_sanity_basic , which is reverted to a commit before refactoring happened. I see sanity check passed. So, I think the refactoring is indeed increasing peak memory usage in sanity checks.
- I am trying the reproduce the memory issue on my local machine when compiled with the same flags as sanity tests and run basic regression tests. But memory usage looks normal on my side. 

Here is part of the log files on Github runner for the sanity test: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1805/checks?check_run_id=3118949531
 
which shows the peak memory usage (=6600MB) exceeds the limit:

```
2021-07-20T23:44:33.0545521Z 	<Router Stats: total_nets_routed: 2643 total_connections_routed: 5927 total_heap_pushes: 1375436 total_heap_pops: 330402
2021-07-20T23:44:33.0546304Z 	<## Build routing resource graph
2021-07-20T23:44:33.0551765Z 	<## Build routing resource graph took 4.21 seconds (max_rss 6161.6 MiB, delta_rss +616.5 MiB)
2021-07-20T23:44:33.0552496Z 	<  RR Graph Nodes: 12681
2021-07-20T23:44:33.0556763Z 	<  RR Graph Edges: 64556
2021-07-20T23:44:33.0562290Z 	<Best routing used a channel width factor of 52.
2021-07-20T23:44:33.0563186Z 	<# Routing took 185.82 seconds (max_rss 6238.1 MiB, delta_rss +5650.7 MiB)
2021-07-20T23:44:33.0566828Z 	<
2021-07-20T23:44:33.0567416Z 	<Checking to ensure routing is legal...
2021-07-20T23:44:33.0571623Z 	<# Checking to ensure non-configurable edges are legal
2021-07-20T23:44:33.0577702Z 	<# Checking to ensure non-configurable edges are legal took 0.18 seconds (max_rss 6314.4 MiB, delta_rss +26.4 MiB)
2021-07-20T23:44:33.0578769Z 	<Completed routing consistency check successfully.
2021-07-20T23:44:33.0579369Z 	<
2021-07-20T23:44:33.0585087Z 	<Serial number (magic cookie) for the routing is: -823726116
2021-07-20T23:44:33.0585850Z 	<Circuit successfully routed with a channel width factor of 52.
2021-07-20T23:44:33.0592363Z 	<# Synchronize the packed netlist to routing optimization
2021-07-20T23:44:33.0593104Z 	<## Annotating rr_node with routed nets
2021-07-20T23:44:33.0593826Z 	<## Annotating rr_node with routed nets took 0.00 seconds (max_rss 6315.8 MiB, delta_rss +0.6 MiB)
2021-07-20T23:44:33.0600922Z 	<Found 1212 mismatches between routing and packing results.
2021-07-20T23:44:33.0601978Z 	<Fixed 878 routing traces due to mismatch between routing and packing results.
2021-07-20T23:44:33.0607271Z 	<# Synchronize the packed netlist to routing optimization took 1.01 seconds (max_rss 6462.1 MiB, delta_rss +146.9 MiB)
2021-07-20T23:44:33.0612892Z 	<Final Clustering Statistics: 
2021-07-20T23:44:33.0616380Z 	<----------   --------   ------------------------------------   --------------------------
2021-07-20T23:44:33.0619430Z 	<Block Type   # Blocks   Avg. # of input clocks and pins used   Avg. # of output pins used
2021-07-20T23:44:33.0622962Z 	<----------   --------   ------------------------------------   --------------------------
2021-07-20T23:44:33.0626564Z 	<     EMPTY          0                                      0                            0   
2021-07-20T23:44:33.0629612Z 	<        io        258                               0.372093                     0.627907   
2021-07-20T23:44:33.0632595Z 	<       clb         72                                13.7222                      4.80556   
2021-07-20T23:44:33.0633303Z 	<    memory          0                                      0                            0   
2021-07-20T23:44:33.0634001Z 	<   mult_36          5                                   65.6                         33.6   
2021-07-20T23:44:33.0638354Z 	<Absorbed logical nets 538 out of 1214 nets, 676 nets not absorbed.
2021-07-20T23:44:33.0638888Z 	<
2021-07-20T23:44:33.0642641Z 	<
2021-07-20T23:44:33.0643447Z 	<Average number of bends per net: 2.53640  Maximum # of bends: 38
2021-07-20T23:44:33.0644076Z 	<
2021-07-20T23:44:33.0649409Z 	<Number of global nets: 3
2021-07-20T23:44:33.0650024Z 	<Number of routed nets (nonglobal): 673
2021-07-20T23:44:33.0656348Z 	<Wire length results (in units of 1 clb segments)...
2021-07-20T23:44:33.0657031Z 	<	Total wirelength: 9173, average net length: 13.6300
2021-07-20T23:44:33.0663225Z 	<	Maximum net length: 162
2021-07-20T23:44:33.0663647Z 	<
2021-07-20T23:44:33.0667094Z 	<Wire length results in terms of physical segments...
2021-07-20T23:44:33.0667864Z 	<	Total wiring segments used: 2608, average wire segments per net: 3.87519
2021-07-20T23:44:33.0677146Z 	<	Maximum segments used by a net: 48
2021-07-20T23:44:33.0677820Z 	<	Total local nets with reserved CLB opins: 0
2021-07-20T23:44:33.0678326Z 	<
2021-07-20T23:44:33.0681247Z 	<Routing channel utilization histogram:
2021-07-20T23:44:33.0681827Z 	<[        1:      inf)   0 (  0.0%) |
2021-07-20T23:44:33.0682302Z 	<[      0.9:        1)   6 (  2.1%) |***
2021-07-20T23:44:33.0689095Z 	<[      0.8:      0.9)  66 ( 22.9%) |******************************
2021-07-20T23:44:33.0689674Z 	<[      0.7:      0.8) 104 ( 36.1%) |***********************************************
2021-07-20T23:44:33.0690145Z 	<[      0.5:      0.6)  42 ( 14.6%) |*******************
[... cut ...]
2021-07-20T23:44:33.0844511Z 	<Segment usage by length: length utilization
2021-07-20T23:44:33.0848388Z 	<                         ------ -----------
2021-07-20T23:44:33.0849114Z 	<                              4       0.597
2021-07-20T23:44:33.0849555Z 	<
2021-07-20T23:44:33.0854472Z 	<Flow timing analysis took 0 seconds (0 STA, 0 slack) (0 full updates: 0 setup, 0 hold, 0 combined).
2021-07-20T23:44:33.0855162Z 	<VPR succeeded
2021-07-20T23:44:33.0859699Z 	<The entire flow of VPR took 306.66 seconds (max_rss 6602.1 MiB)
```

tangxifan, 2021-07-29:
Here are the log files from my local runs on the sanity basic tests:

[basic_no_timing_ch_intrinsics_vpr.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6897456/vpr_stdout.log)
[basic_no_timing_diffeq1_vpr.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6897460/vpr_stdout.1.log)
[basic_timing_ch_intrinsics_vpr.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6897462/vpr_stdout.2.log)
[basic_timing_ch_intrinsics_reordered_vpr.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6897464/vpr_stdout.3.log)
[basic_timing_diffeq1_vpr.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6897466/vpr_stdout.4.log)
[basic_timing_diffeq1_reordered_vpr.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6897468/vpr_stdout.5.log)

vaughnbetz, 2021-07-29:
Thanks @tangxifan. That is a huge memory footprint increase; from ~300 MB to 6.6 GB or so then?

tangxifan, 2021-07-29:
Yes. On the same test case ``basic_no_timing_diffeq1``, the local run log shows 250MB peak memory usage, while on the github runner it shows 6.6 GB.

sfkhalid, 2021-07-29:
I re-ran the tests on #1802 today and so far of two of the tests that had been failing are passing (Test / S: Basic (push) and Test / ODIN-II Basic Tests (push))

hzeller, 2021-07-29:
Was it ruled out that it could have something to do with large memory pages, which could be a configuration on the github machines in question ?
Wild speculation, but sanitizers often work by placing allocations near page lines to detect things automatically by triggering page faults on overruns.

vaughnbetz, 2021-07-29:
I don't believe that has been ruled out.  Not sure how to check the configuration of the github runners though, and if they were changed.

tangxifan, 2021-07-29:
> I re-ran the tests on #1802 today and so far of two of the tests that had been failing are passing (Test / S: Basic (push) and Test / ODIN-II Basic Tests (push))

Same on my side: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1805

sdamghan, 2021-07-29:
I had an issue with Github runner about a couple of weeks ago regarding a value in Yosys+Odin's expected outputs. The expected result generated in my Desktop (Ubuntu 20.04) was different from the one GitHub runner produced for a single value among all matched values. I ran tests on CentOS and the result matched. Although the git runner (`'2.279.0'`) OS is `Ubuntu 18.04.5 LTS`, I have generated the Odin outputs for failed benchmarks in Odin reg basic in both Ubuntu and CentOS (the same as git runner config for CMake). The maximum held memory was `1867.8` in Ubuntu and `2171.3` in CentOS, which is way less than the git runner memory limit. Not sure why Odin reg basic fails, but obviously, some Odin benchmarks fail because of the memory limit, while they passed this test previously without any change.

Ubuntu:
         -  [(cf_fft_1024_16)_(no_arch)_(Ubuntu_20.04).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898285/cf_fft_1024_16._.no_arch._.Ubuntu_20.04.log)
         - [(ex2PM16_fir_119)_(k6_N10_40nm)_(Ubuntu_20.04).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898286/ex2PM16_fir_119._.k6_N10_40nm._.Ubuntu_20.04.log)
         - [(ex4LS16_fir)_(k6_N10_40nm)_(Ubuntu_20.04).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898288/ex4LS16_fir._.k6_N10_40nm._.Ubuntu_20.04.log)
         - [(ex4LS16_fir)_(k6_N10_mem32K_40nm)_(Ubuntu_20.04).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898289/ex4LS16_fir._.k6_N10_mem32K_40nm._.Ubuntu_20.04.log)
         - [(ex4PM16_fir_152)_(no_arch)_(Ubuntu_20.04).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898290/ex4PM16_fir_152._.no_arch._.Ubuntu_20.04.log)
 
 CentOS:
         - [(cf_fft_1024_16)_(no_arch)_(CentOS).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898340/cf_fft_1024_16._.no_arch._.CentOS.log)
         - [(ex2PM16_fir_119)_(k6_N10_40nm)_(CentOS).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898341/ex2PM16_fir_119._.k6_N10_40nm._.CentOS.log)
         - [(ex4LS16_fir)_(k6_N10_40nm)_(CentOS).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898342/ex4LS16_fir._.k6_N10_40nm._.CentOS.log)
         - [(ex4LS16_fir)_(k6_N10_mem32K_40nm)_(CentOS).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898343/ex4LS16_fir._.k6_N10_mem32K_40nm._.CentOS.log)
         - [(ex4PM16_fir_152)_(no_arch)_(CentOS).log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6898344/ex4PM16_fir_152._.no_arch._.CentOS.log)

ArashAhmadian, 2021-07-29:
> I had an issue with Github runner about a couple of weeks ago regarding a value in Yosys+Odin's expected outputs. The expected result generated in my Desktop (Ubuntu 20.04) was different from the [...]

I was looking at some of the SIGABRTs earlier today and they seem to be Odin aborting due to e.g. not supported syntax, instantiating non-existent hard-blocks, etc. They don't seem to be actual failures, but expected behavior from Odin since the golden_results for simulation/synthesis match the produced results. Is this understanding somewhat correct? @sdamghan   

There could be ones that go over the memory limit, however none that I've the checked the `simulation.log` and/or `synthesis.log` for indicate this.

sdamghan, 2021-07-29:
@ArashAhmadian yes actually, those are showing SIGABRT are expected to fail. Current failures are for some benchmarks that overuse memory, while they do not use that much memory in local machines (even with sanitizer on). I suspected the issue might come from the runner OS; however, it seemed it was not our case.

sdamghan, 2021-07-29:
> @sdamghan : this PR #1807 has a strange behaviour. It appears Odin reg basic succeeded (which would be odd given the sanitization issues in CI) but when I look at the logs I see multiple sigabrts in [...]

@vaughnbetz PR #1807 _not failure_ status   was caused by the verification script. For VTR benchmarks the return value was not considered properly. The solution is pushed in PR #1807 
Also in two PRs #1798 and #1807 that I have recently updated, there are no `Odin reg basic` or `Vtr reg strong` failures caused by memory limit!

vaughnbetz, 2021-07-29:
Thanks for the detailed investigation @sdamghan and @ArashAhmadian.  That's good news the SIGABRTs are expected, and thanks for fixing the return value.  At least we have only one problem now: why github runners take so much memory with sanitizers on.

mithro, 2021-07-29:
@kgugala - Could you help by getting someone with GitHub Actions experience from Antmicro to take a take a look at this?

vaughnbetz, 2021-07-29:
@sdamghan : could you change the output from the Odin II regtest log to show "SIGABRT (expected)" or some such to indicate expected failures?  Or have the log file otherwise print some lines explaining that SIGABRT is a valid output, so long as it matches the expected output.  I imagine we may have others confused in the future by the log (as I was) thinking the SIGABRT status for various tests indicated a problem.

sdamghan, 2021-07-29:
> @sdamghan : could you change the output from the Odin II regtest log to show "SIGABRT (expected)" or some such to indicate expected failures? Or have the log file otherwise print some lines [...]

@vaughnbetz sure, I will add a comprehensive message explaining if any failure is expected.

tangxifan, 2021-08-23:
@vaughnbetz This issue seems to be resolved last week. But I think we can continue the discussion on enabling artifacts on our CI runner. It will definitely ease our QoR checks for VTR benchmarks, Titan benchmarks as well as other crucial testing suites we care.

We consider the following improvements
- Enable the [run_reg_test.py](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/run_reg_test.py) script to optionally show peak memory usage in the print out. This will be very useful when checking any memory leak on our sanity tests. For example

```
==========================================================================
                  Verilog-to-Routing Regression Testing
==========================================================================
          Running vtr_reg_basic
------------------------------------------
scripts/run_vtr_task.py -l /home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/vtr_flow/tasks/regression_tests/vtr_reg_basic/task_list.txt -j 2 -script run_vtr_flow.py -show_failures -short_task_names 

k6_N10_mem32K_40nm/mkPktMerge		OK (took 1431.56 seconds, consumed 400MB memory)
```

- Enable artifacts to be downloaded on Github runners. This will ease us to check and debug  any QoR results on any VTR tests deployed on Github CI system. For example, in OpenFPGA project, we have tested this feature.

[image]

- Enable artifacts to be downloaded on Kokoro runner. This will definitely ease our QoR check on VTR tests. As discussed before, @kgugala may take the lead on this effort. We would like to help as well.

acomodi, 2021-09-02:
It is unclear actually what needs to be done regarding the artifacts on Kokoro. When Kokoro completes the tasks, the artifacts are uploaded to GCS, and those include all the QoR data to be inspected.

umarcor, 2021-09-02:
> @umarcor : did the removal of travis in #1804 lead to any configuration settings that have already been changed that might have caused this?

Sorry about the delay in getting back here. I tried to be as strict as possible in preserving the previous behaviour throughout my PRs. Since I'm not confident with the codebase in this repository, I did not want to make any assumption. That lead to some "weird" refactorisation outcomes, such as #1813. Naturally, I am human, and I do silly mistakes, but there was no conscious modification at all.

umarcor, 2021-09-02:
> @vaughnbetz This issue seems to be resolved last week.

@tangxifan was it solved by itself or did you actually change anything in this repo?

vaughnbetz, 2021-09-02:
It seemed like it was a change on the github side (something in their machine setups?) as the out-of-memory problem stopped occurring without us making a change. Thanks for the follow up though @umarcor !

tangxifan, 2021-09-02:
> It is unclear actually what needs to be done regarding the artifacts on Kokoro. When Kokoro completes the tasks, the artifacts are uploaded to GCS, and those include all the QoR data to be inspected.

@acomodi Thanks for the inputs. What we are looking for is the QoR data. If the current set-up can allow us to download the QoR report file (.txt or .csv), it is already good enough. If you can show me how to access these files, I think the problem is solved here.

acomodi, 2021-09-02:
@tangxifan I think that here the GCS API can be used to script the data gathering.

In general, to access the GCS bucket, you can navigate to `Invocation Details -> GCS (link)` from a kokoro run. There you can than navigate to see all the various results of the various benchmarks.

One possible thing to make this step easier might be to just package all of the QoR txt files in a single tarball, so that everything can be downloaded once only for each kokoro run.

tangxifan, 2021-09-02:
@acomodi Thanks a lot! I managed to find out all the logs I need. It is indeed solves my issue perfectly.

I attached the screenshot 

[image]

mithro, 2022-02-24:
@acomodi / @tangxifan / @ganeshgore -- Is this bug still relevant?

umarcor, 2022-02-24:
I think this can be closed. The dialogue drifted towards "which artifacts to save", and that was handled as well.

ganeshgore, 2022-02-24:
Yes, this issue is already addressed.

closed 2022-02-24 by mithro