verilog-to-routing/vtr-verilog-to-routing #1815: "num_pins > 0" assertion fails during alloc_and_load_pb_graph_pin_lookup_from_index

TaoBi22, 2021-08-02:
Hi,

I'm working on the FABulous project (https://github.com/FPGA-Research-Manchester/FABulous), aiming to add functionality to program our generated FPGA fabrics with VTR. I've come across an odd issue testing with our generated architecture descriptions, which I can't seem to find record of elsewhere, so any help anyone could give would be very appreciated.

Running with the following command "./run_vtr_flow.pl and_gate.v ../../../FABulous/fabric_generator/vproutput/architecture.xml -lut_size 4 --route_chan_width 2", and_gate.v being this simple circuit:

```
module and_gate(
	a_in,
	b_in,
	out
);

    // SIGNAL DECLARATIONS
    input	a_in, b_in;

    output	out;

    // ASSIGN STATEMENTS
    always @ (*)
    begin
        out = a_in & b_in;
    end

endmodule
```

I receive the message "failed: vpr (exited with return code 134) (took 0.33 seconds)". Looking into temp/vpr_stdout.log it seems that this is due to an assertion that num_pins > 0 has failed during alloc_and_load_pb_graph_pin_lookup_from_index:

[vpr_stdout.log](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6916390/vpr_stdout.log)

I'm also unsure why this logfile seems to report that there are insufficient resources, as the circuit should obviously be very simple, so this may be related (I can't actually track down why the condition for this being logged is being evaluated as true, as checking the values it tests directly after evaluation, it seems to evaluate to 1 > 48, but I'm obviously not aware of the details of this process).

This issue is occurring with VPR 8.0.0 on an Arch Linux machine (compiled with GCC), and the architecture.xml file is below (I've swapped out the file extension for .txt so that it can be attached)

[architecture.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/6916414/architecture.txt)

There are a couple of things that may seem unusual about the file, but hopefully are not cause for concern:
- We intend to provide import a custom routing resource graph, hence the lack of routing architecture (however, including this file with the read_rr_graph flag does not change the output, so I thought it best to leave it out in this issue for simplicity's sake)
- The clock within the LUT4 complex block is not connected to anything - this is a known issue and the clock specification is still being worked on, but again, adding dummy top-level interconnections does not change the output so this does not seem to be related to this issue.
- Currently, on automatically generated models, all inputs have all outputs as their combinational sink ports - this is just a filler for now so we can get most of the generation done before involving a more complex timing model.

We've been puzzling over this for a few days and had a look through the underlying code but we're struggling to find the root of the problem, so if anyone has any idea what could be causing this, it would be greatly appreciated - I believe I've attached/explained everything relevant but let me know if I've missed anything! Thanks!

vaughnbetz, 2021-08-12:
The "insufficient resources" message isn't a problem; you're running with an auto-sized fpga (the default), so a larger grid (chip) is created whenever it's clear the packer needs more resources.  So just consider those progress messages.

I'm also not sure of what's causing the assertion.  The first thing I'd try is the latest (master) code to see if the issue is already resolved though.  There have been quite a few changes from V8.0.0 (the 8.0 release).  You can just check out the master from the vtr repo and try it.

TaoBi22, 2021-08-16:
That's good to know, thanks!!

I'm trying to build master at the moment and having a couple of issues with my system but I'll report back once I've managed to get it up and running. A recent development is that adding an extra input port (to add another feature) to every tile and top-level pb_type seems to have stopped the assertion failing for some reason, but I'll see what happens on master with the original file.

TaoBi22, 2021-08-18:
I adjusted the file slightly to meet Odin II's sub_tile requirements that have been added since 8.0.0, ran again and it looks like the assertion is still failing - the new architecture file and vpr.out are attached below.

[architecture.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/7006471/architecture.txt)
[vpr.out.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/7006477/vpr.out.txt)

However, since the assertion is no longer failing with my most recent code (although I'm not too sure why!), and I assume I'm the first to experience this problem, I'm happy to assume it was a mistake in my architecture description that slipped under the radar and close the issue, unless you think it merits further investigation?

vaughnbetz, 2021-08-20:
That seeM’s reasonable to me. Thanks for the follow up.

closed 2021-08-23 by TaoBi22