verilog-to-routing/vtr-verilog-to-routing #1823: Runs are now failing Python Lint test in github actions CI

heinerb1, 2021-08-09:
After PR #1793 which dropped python 2.7 was merged into master yesterday, all runs are now failing the Python Lint test. According to the output, this test is failing due to lint errors in vtr_flow/scripts/run_vtr_task.py and no other files seem to have been affected.

mithro, 2021-08-09:
```
vtr_flow/scripts/run_vtr_flow.py passed 
vtr_flow/scripts/run_vtr_task.py has lint errors 
************* Module run_vtr_task
vtr_flow/scripts/run_vtr_task.py:240:37: E1136: Value 'common_task_prefix' is unsubscriptable (unsubscriptable-object)
vtr_flow/scripts/download_symbiflow.py passed 
vtr_flow/scripts/arch_gen/arch_gen.py skipped (grandfathered) 
vtr_flow/scripts/python_libs/vtr/task.py passed 
vtr_flow/scripts/python_libs/vtr/error.py passed 
vtr_flow/scripts/python_libs/vtr/util.py passed 
vtr_flow/scripts/python_libs/vtr/__init__.py passed 
vtr_flow/scripts/python_libs/vtr/flow.py passed 
vtr_flow/scripts/python_libs/vtr/parse_vtr_task.py passed 
vtr_flow/scripts/python_libs/vtr/log_parse.py passed 
vtr_flow/scripts/python_libs/vtr/paths.py passed 
vtr_flow/scripts/python_libs/vtr/parse_vtr_flow.py passed 
vtr_flow/scripts/python_libs/vtr/abc/abc.py passed 
vtr_flow/scripts/python_libs/vtr/abc/__init__.py passed 
vtr_flow/scripts/python_libs/vtr/ace/ace.py passed 
vtr_flow/scripts/python_libs/vtr/ace/__init__.py passed 
vtr_flow/scripts/python_libs/vtr/vpr/__init__.py passed 
vtr_flow/scripts/python_libs/vtr/vpr/vpr.py passed 
vtr_flow/scripts/python_libs/vtr/odin/__init__.py passed 
vtr_flow/scripts/python_libs/vtr/odin/odin.py passed 
vtr_flow/scripts/spice/run_spice.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/populate_db.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/plotter-offline.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/util.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/__init__.py passed 
vtr_flow/scripts/benchtracker/server_db.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/interface_db.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/flask_cors/six.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/flask_cors/__init__.py passed 
vtr_flow/scripts/benchtracker/flask_cors/version.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/flask_cors/decorator.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/flask_cors/extension.py skipped (grandfathered) 
vtr_flow/scripts/benchtracker/flask_cors/core.py skipped (grandfathered) 
ERROR: 1 file(s) failed lint test. 
Error: Process completed with exit code 1.
```

vaughnbetz, 2021-08-12:
This was fixed in PR #1821

closed 2021-08-12 by vaughnbetz