verilog-to-routing/vtr-verilog-to-routing #1832: Request for a Kokoro Runner

sdamghan, 2021-08-17:
#### Context

@mithro Would you please assign a new Kokoro runner for vtr_reg_yosys+odin? [[presubmit](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/d779c50137f3ef42a84dae438198531748b64847/.github/kokoro/presubmit/yosys%2Bodin_test.cfg)] [[continuous](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/d779c50137f3ef42a84dae438198531748b64847/.github/kokoro/continuous/yosys%2Bodin_test.cfg)]

mithro, 2021-08-17:
@QuantumHD - Can you do this for me? (Happy to show you how.)

QuantamHD, 2021-08-18:
Yeah! Just let me know what I should do

QuantamHD, 2021-08-18:
@sdamghan Can you remove the `+` symbols from the file names?

sdamghan, 2021-08-18:
@QuantamHD thanks for following up, I’ll rename Kokoro config files in the next commits and inform you.

sdamghan, 2021-08-23:
@QuantamHD I have just updated the Kokoro config file names; please let me know if you need further modifications.