verilog-to-routing/vtr-verilog-to-routing #1846: Load rr_graph too slow in windows

the-centry, 2021-09-13:
I compile the latest vpr with cygwin and vs2019 ,however I found that it was too slow while loading rr_graph in windows; In linux I load the same rr_graph took about 60s but it need 3600s in windows! It's the problem of capnp? 
  The rr_graph generated by [symbiflow](https://github.com/SymbiFlow/symbiflow-arch-defs).
  Is there some way to load rr_graph quickly in windows?
  Thanks so much!

zhugesima, 2021-11-12:
Have you solved this problem? If it is solved, can you share your solution? Thanks very much!

the-centry, 2021-12-27:
> Have you solved this problem? If it is solved, can you share your solution? Thanks very much!

I just changed some computes and found the results are different!

closed 2021-12-27 by the-centry

aa6my, 2022-11-21:
@the-centry may i know how you compile vpr with cygwin and vs2019 ?